WebODM/WebODM #126: Add KML to downloadable assets for the Orthomosaic

basrijn, 2017-03-21:
Loading the produced orthomosaic in Google Earth is a great way to review the output (and georeferencing accuracy). Most people have the standard Google Earth that does not support the direct loading of GeoTIFF's.

The addition of a KML file in the Assets download section would be very helpful.

It seems that gdal might already have this functionality (via gdal2tiles, https://developers.google.com/kml/articles/raster)

Bas

smathermather, 2017-03-21:
Excellent idea. @pierotofy & @dakotabenjamin -- do you think this should be WebODM only, or should we put it in the upstream ODM?

pierotofy, 2017-03-22:
I'd be cool if this was implemented upstream and then WebODM simply imported the resulting assets.

smathermather, 2017-03-22:
Easy enough. Cool.

pierotofy, 2017-05-02:
@basrijn are you still interested in implementing this feature? No pressure, just wanted to check the status 😄

atoone, 2017-05-17:
This would really help sharing orthophotos with others...

fentas, 2020-05-26:
Has anyone begun to work on this? :cat2:

pierotofy, 2020-05-26:
Nop. Still looking for a contribution if you're intetested.

pierotofy, 2020-06-03:
We can now raise funds to get this done using quadratic funding! https://fund.webodm.org/#/fund/2

pierotofy, 2020-07-04:
Closing this as I don't think it's going to get implemented (unless we get a pull request).

closed 2020-07-04 by pierotofy