WebODM/WebODM #134: Service 'db' failed to build

zengjt, 2017-03-29:
run " ./webodm.sh start" , then  errors appeared:

E: Failed to fetch http://deb.debian.org/debian/pool/main/libs/libsamplerate/libsamplerate0_0.1.8-8_amd64.deb  Size mismatch

E: Failed to fetch http://deb.debian.org/debian/pool/main/f/faad2/libfaad2_2.7-8_amd64.deb  Size mismatch

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: Service 'db' failed to build: The command '/bin/sh -c apt-get update       && apt-get install -y --no-install-recommends            postgresql-$PG_MAJOR-postgis-$POSTGIS_MAJOR            postgresql-$PG_MAJOR-postgis-$POSTGIS_MAJOR-scripts            postgis       && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100

pierotofy, 2017-03-29:
Hello @zengjt, for some reason your docker environment is not able to pull those two debian packages. I've just run `docker-compose build --no-cache db` on my machine and it seemed to work OK, so I'm guessing it's probably related to your network configuration, or perhaps deb.debian.org was down when you tried to start WebODM. Things to try:

1. Launch `./webodm.sh start` again
2. Check your network configuration, make sure deb.debian.org is accessible, make sure your traffic is not being filtered, that you are not connected to a proxy/VPN

The suggestion to run `apt-get update` does not apply, because we already do that in the dockerfile as part of the build process.

zengjt, 2017-03-30:
pierotofy，thanks for your help. But I downloaded the file  with the size  937.4 kB from "http://deb.debian.org/debian/pool/main/libs/libsamplerate/libsamplerate0_0.1.8-8_amd64.deb", 
so I think it is not related to  network configuration.

pierotofy, 2017-03-30:
http://askubuntu.com/questions/165604/getting-size-mismatch-when-using-apt-get-on-multiple-packages

pierotofy, 2017-03-30:
This is the filesize in my machine:

`-rw-r--r-- 1 piero piero 948832 Mar 30 08:24 libsamplerate0_0.1.8-8_amd64.deb`

Or 927K.

pierotofy, 2017-04-03:
Will close this unless there's more questions.

closed 2017-04-03 by pierotofy

zengjt, 2017-04-04:
i modify the Dockerfile in 'db' folder with 'postgres:9.5' replaced with 'postgres:9.6',   but the same errors appeared. I found current python version is 2.7 ,  so i change it to 3.5.  But new errors appeared as follows:
Checking for docker...   OK
Checking for git...   OK
Checking for python...   OK
Checking for pip...   OK
Checking for docker-compose...   OK
Starting WebODM...
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Traceback (most recent call last):
  File "/usr/local/bin/docker-compose", line 7, in <module>
    from compose.cli.main import main
ImportError: No module named 'compose'

pierotofy, 2017-04-04:
You shouldn't need to modify postgres version, or python (2.7 should work just fine).

Like I mentioned earlier, either something on your network or on your ISP is tampering with your downloads.

zengjt, 2017-04-04:
I used ubuntu16.04 in VMware environment in win7. Aim to the error 'size mismatch', I tried to install postgres9.5 natively and succeeded in that. In addition, I  tried to use lantern as a vpn  to run './webodm.sh start'  and got the same errors finally.  I found  it was successful of  fetching lib files from 'http://deb.debian.org/debian/ jessie/....'  whereas failed from ' http://deb.debian.org/debian/pool/...',  and the latter  is  for lib files of  'size mismatch' errors