WebODM/WebODM #135: Run WebODM natively

paulinK, 2017-03-31:
Hello
I'am a ftench speaker.
I'am new on WebODM. I have try to run WebODM nativily but it failed. And I don't know why. My question is that:
Can I get a tutorial about running WebODM natively ?

pierotofy, 2017-03-31:
Hi @paulinK,

did you try to follow the instructions written here? https://github.com/OpenDroneMap/WebODM#run-it-natively

paulinK, 2017-03-31:
Thank you @pierotofy .
But it dosen't work . i triée it

paulinK, 2017-03-31:
Can I get another link ?
Thank you

pierotofy, 2017-03-31:
@paulinK where exactly do you get stuck in the process? There's no other documentation at the moment.

paulinK, 2017-03-31:
Ah ok. I will post thé errors when i will on my computer.
Thank you again.

pierotofy, 2017-04-05:
Will close this unless there are further questions.

closed 2017-04-05 by pierotofy