WebODM/WebODM #136: node-odm-1 exited with code 1

gkconsulting, 2017-03-31:
Hi, 
I have been trying to get Web ODM to run on a windows machine with Docker. 
I have followed the instructions and I can get into the interface however then I try to set a processing node I get a message saying "node-odm-1:3000 seems to be offline. "
On the terminal i get this error "error:Error during startup: Could not load list of options from OpenDroneMap. Is OpenDroneMap installed in /code? Make sure that OpenDroneMap is installed and that --odm-path is set properly: Unexpected token T in JSON at position 0"

Is there a step that I am missing? I have followed the instructions on this link "https://github.com/OpenDroneMap/WebODM"
Thanks in advance.
GK

pierotofy, 2017-03-31:
Hi, our docker builds were broken yesterday due to a major update. Could you please try to run the following:

```
./webodm.sh update
./webodm.sh start
```

And see if that fixes the problem?

gkconsulting, 2017-03-31:
Thanks, 
Will give it a try and let you know. 
Thanks so much for the quick response
GK

gkconsulting, 2017-03-31:
Thanks that did the trick. 
Thanks for all the hard work :-)

closed 2017-03-31 by gkconsulting

pierotofy, 2017-03-31:
Glad it worked! 😄 

Feel free to give us feedback once you are done checking out the project.