WebODM/WebODM #137: Failed to execute script docker-compose

jdzurisin, 2017-04-04:
Attempting to get WebODM up and running.

Windows 7 Pro
Installed Docker Toolbox
Installed Python 3.5.0
Installed Git

When running ./webodm.sh start I get:

File "site-packages\docker\transport\npipeconn.py", line 50, in connect
pywintypes.error: (2, 'WaitNamedPipe', 'The system cannot find the file specified.')
Failed to execute the script docker-compose

Googled myself into a corner. Any help you might be able to provide would be greatly appreciated.

pierotofy, 2017-04-04:
Hi @jdzurisin thanks for reporting the issue.

Are you running ./webodm.sh start in the Docker Quickstart Terminal?

Also, try to downgrade Python to 2.7 and see if that fixes it.

jdzurisin, 2017-04-04:
Solved. Needed to update BIOS to enable VT-x virtualization.

pierotofy, 2017-04-04:
I'll add a note in the `README`. Thanks for reporting the solution!

closed 2017-04-04 by pierotofy

stiward, 2017-08-03:
Hello, this solution did not work, download Python 2.7 and continue giving me the same error, my virtual machine is running because but Docker will not work.

Please I need help !!!
``Starting webapp ... error

ERROR: for db  (2, 'WaitNamedPipe', 'The system cannot find the file specified.')

ERROR: for node-odm-1  (2, 'WaitNamedPipe', 'The system cannot find the file specified.')
Traceback (most recent call last):
  File "docker-compose", line 3, in <module>
  File "compose\cli\main.py", line 68, in main
  File "compose\cli\main.py", line 118, in perform_command
  File "compose\cli\main.py", line 792, in start
  File "compose\project.py", line 247, in start
  File "compose\parallel.py", line 69, in parallel_execute
pywintypes.error: (2, 'WaitNamedPipe', 'The system cannot find the file specified.')
Failed to execute script docker-compose
Traceback (most recent call last):
  File "docker-compose", line 3, in <module>
  File "compose\cli\main.py", line 68, in main
  File "compose\cli\main.py", line 118, in perform_command
  File "compose\cli\main.py", line 926, in up
  File "compose\project.py", line 388, in up
  File "compose\project.py", line 614, in warn_for_swarm_mode
  File "site-packages\docker\api\daemon.py", line 90, in info
  File "site-packages\docker\utils\decorators.py", line 47, in inner
  File "site-packages\docker\api\client.py", line 183, in _get
  File "site-packages\requests\sessions.py", line 488, in get
  File "site-packages\requests\sessions.py", line 475, in request
  File "site-packages\requests\sessions.py", line 596, in send
  File "site-packages\requests\adapters.py", line 423, in send
  File "site-packages\requests\packages\urllib3\connectionpool.py", line 595, in urlopen
  File "site-packages\requests\packages\urllib3\connectionpool.py", line 363, in _make_request
  File "httplib.py", line 1042, in request
  File "httplib.py", line 1082, in _send_request
  File "httplib.py", line 1038, in endheaders
  File "httplib.py", line 882, in _send_output
  File "httplib.py", line 844, in send
  File "site-packages\docker\transport\npipeconn.py", line 31, in connect
  File "site-packages\docker\transport\npipesocket.py", line 22, in wrapped
  File "site-packages\docker\transport\npipesocket.py", line 50, in connect
pywintypes.error: (2, 'WaitNamedPipe', 'The system cannot find the file specified.')
Failed to execute script docker-compose``