WebODM/WebODM #138: ERROR: for webapp

ITPrincis, 2017-04-05:
Hello,

I tried to start WebODM like Dennis shown in youtube video, however I ran into some problems:

terminal shows:
ERROR: for webapp  No such image: sha256:c7046a00c84fb4fc2f7d8faa9ebba55d88e7ae813e5e3dbf0231bbd964ba9df0
ERROR: Encountered errors while bringing up the project.
[image]

pierotofy, 2017-04-05:
Hi @ITPrincis , thanks for reporting the issue.

Have you ever been able to launch webodm? 

I would try to do a `docker-compose build --no-cache webapp` from the WebODM directory and try again.

ITPrincis, 2017-04-05:
Hi,

no I'm new to this. I haven't launched it yet, but I'm about to do it.
Ok, so it created webodm_webapp image.

How can I run it?
Where can I see what ip is local server running on?

Big thanks!
[image]

ITPrincis, 2017-04-05:
@pierotofy 
When I run ./devenv.sh start
then this shows again:
[image]

pierotofy, 2017-04-05:
Something does not look right. I'm not sure what commands you've typed, but if you started from a fresh installation, all you need to type is:

```
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input
cd WebODM
./webodm.sh start
```

I would try to clean things up and start over...

```
docker stop $(docker ps -q)
docker rm $(docker ps -aq)
./webodm.sh rebuild
./webodm.sh start
```

You don't need to use `devenv.sh` unless you plan to edit the source code.

ITPrincis, 2017-04-05:
@pierotofy Started fresh and everything went well. No errors. Thanks!

closed 2017-04-05 by ITPrincis