WebODM/WebODM #140: Issue connecting browser to WebODM

AnthonyCoughlin, 2017-04-08:
@pierotofy 

Long time Sir.  When I start WebODM via Docker then attempt to open via browser its not connecting.  I'm sure I'm missing something simple here...

Thanks,
Anthony

[image]

[image]

pierotofy, 2017-04-08:
Hello Anthony,

What is the output of `docker-machine ip`?

Also perhaps an obvious thing to try, but have you tried to restart your computer?

AnthonyCoughlin, 2017-04-08:
@pierotofy 

see below.  Yes restarted both the machine and the docker.

[image]

pierotofy, 2017-04-08:
I found this while searching online. http://www.hanselman.com/blog/FixedMicrosoftEdgeCantSeeOrOpenVirtualBoxhostedLocalWebSites.aspx

Does it help at all your case?

AnthonyCoughlin, 2017-04-08:
Hmmm, testing Firefox now.  Thanks.

AnthonyCoughlin, 2017-04-08:
@pierotofy 

Saved my life again my friend.  I was running the all powerful Microsoft Edge and it would not connect.

pierotofy, 2017-04-08:
Glad it worked!

I've added a note in the readme for future reference. Thank you for reporting the problem!

closed 2017-04-08 by pierotofy