WebODM/WebODM #141: exit with code 1

AnthonyCoughlin, 2017-04-09:
@pierotofy 

I'm experiencing the 'exited with code 1' message.  Running a task with 118 photos trying to make a 3D model.  Any ideas?

[image]

AnthonyCoughlin, 2017-04-09:
I'm getting the same errors listed above on three separate data sets.

pierotofy, 2017-04-09:
Hard to say what the problem would be from just the error logs posted; yes please if you could upload the images that would be awesome!

AnthonyCoughlin, 2017-04-09:
@pierotofy 

https://drive.google.com/open?id=0B4Tf0aIvxRLJTnpKOWlsU3ZaN2c

This is what I'm working with now.

AnthonyCoughlin, 2017-04-09:
@pierotofy 

Just received this error.  Its as if its getting hung up...

[image]

Hopefully this helps.

pierotofy, 2017-04-09:
I was able to process the dataset without errors with the default settings.

[image]

[image]

Have you allocated enough memory to your processing node? Is there enough disk space?

AnthonyCoughlin, 2017-04-09:
Man...  Mine is still running.  I've got plenty of disk space.  Would you mind providing a link to documentation on the proper memory allocation for webodm.  Thanks for the help.

AnthonyCoughlin, 2017-04-09:
Is is safe to manipulate the memory settings for the default docker machine from the VirtualBox UI?

smathermather, 2017-04-10:
@AnthonyCoughlin -- modifying the memory settings is definitely recommended. Defaults tend to be too small.

AnthonyCoughlin, 2017-04-10:
@smathermather 

Understood, thanks.  With that said, could you tell me where I can find documentation as to how one would set the memory, processors, etc. per the physical machine specs?  I've been reading but have been unable to find anything definitive.

Thanks again @pierotofy and @smathermather

pierotofy, 2017-04-10:
Documentation for recommended settings is a work in progress. For now, you'll have to to a trial and error approach.

pierotofy, 2017-04-10:
I'll close this for now unless there are further questions.

closed 2017-04-10 by pierotofy