WebODM/WebODM #145: Almost complete, Failed at georeferencing

AnthonyCoughlin, 2017-04-12:
See error below.  After adjusting memory allocation for docker it looks like I almost had success here.  The process failed at georeferencing.  Is this a still a resource allocation issue or something else...  Thanks again guys!!!

Building objmodel:
	Saving model... 
	Saving model... 
Whole texturing procedure took: 224.354s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/opensfm/bundle_r000.out -inputCoordFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_georeferencing/coords.txt -inputFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /var/www/data/0c4327db-edc1-4ce6-a827-98c9fc7b2fac/odm_georeferencing/odm_georeferencing_model_geo.txt

  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

AnthonyCoughlin, 2017-04-12:
Dataset below.  Could this be that the images are two close together?

Thanks again.

https://drive.google.com/open?id=0B4Tf0aIvxRLJTnpKOWlsU3ZaN2c

pierotofy, 2017-04-12:
139 means that the program odm_georef crashed. How does the `odm_texturing/odm_textured_model_geo.obj ` file look like in Meshlab?

AnthonyCoughlin, 2017-04-12:
The process as a whole failed so I was unable to download any of the results.  Unless you are saying that I can navigate to '/webodm/<project name>/om_texturing/odm_textured_model_geo.obj'???  If so, I'll try this evening.

pierotofy, 2017-04-12:
Ah yes, no you won't be able to do that. I'll try to process the dataset sometimes later today and see if I find anything.

pierotofy, 2017-04-13:
I'm getting the same error. I will investigate further and see what I find.

AnthonyCoughlin, 2017-04-14:
Thanks for the help!

pierotofy, 2017-04-14:
So the point cloud is generated properly:

[image]

But the poisson mesh is reconstructed fairly poorly:

[image]

Which I think in turns makes the rest of the generation pipeline fail. In my case it actually errored out
at the mvs texturing stage.

I'm trying to rerun the meshing with octree depth of 12 instead of 9 to see if things get better. I'm currently working on improving the meshing code here https://github.com/OpenDroneMap/OpenDroneMap/pull/542 but it might be a few weeks before things are ready.

pierotofy, 2017-04-14:
@AnthonyCoughlin try to set the "mesh-octree-depth" parameter from the Advanced options tab when you create the Task to `12`. I was able to get a better mesh this way and finish the process.

[image]

I did not get an orthophoto output however.

pierotofy, 2017-04-19:
I'll close this for now unless there are further questions.

closed 2017-04-19 by pierotofy