WebODM/WebODM #146: Using ODM in my PC natively?

ybh0616, 2017-04-12:
Hello! I've used the WebODM using Docker Toolbox.
Thank you for your great solution.

I'm testing the WebODM-based data processing by using Korean national parks' drone images.
I'm a park ranger of Korea National Park Service(in short, KNPS), and also a GIS/RS analyst.

Now I'm using Windows 10, and today I installed the WebODM natively, not using Docker.
After log-in, I found that there is no node processing in there.

I want to add node processing based on my PC, so how do I have to set a hostname & port?
I checked the test node processing, nodeodm.masseranolabs.com:80, and there was no error.

pierotofy, 2017-04-12:
Hello and welcome!

While WebODM can run natively on Windows 10, node-OpenDroneMap cannot. https://github.com/OpenDroneMap/node-OpenDroneMap

You can certainly run a processing node on your local machine, but you'll have to use docker (easiest option). You could setup a virtual machine running Ubuntu and install node-OpenDroneMap there: https://github.com/OpenDroneMap/node-OpenDroneMap#run-it-natively but you won't gain anything compared to docker performance wise.

I hope this helps answer your question.

ybh0616, 2017-04-12:
Okay, I see! Thank you for your rapid response!

smathermather, 2017-04-12:
Mr. Yu,, a pleasure to see you here, @ybh0616! Cheers.

closed 2017-04-12 by pierotofy