WebODM/WebODM #147: Processing stops at Depthmapping (No issue report)

jamesoutdoorrobotics, 2017-04-13:
Greetings! I have a small issue that I had a question about for WebODM and NodeODM. 

Working on processing imagery over 100 photos per instance. Both on WebODM and NodeODM. Both, with image amounts at, or over 100 do not complete their processing tasks and stops processing altogether with no error report. I have attempted to run several instances to narrow down any possible causes with my systems processing capabilities. Tested Pix4D on same system and could process 2135 images with no failure, so I imagine the processor and RAM can handle the heavy workloads for extended time periods. Does this has to do with restrictions in the current application stack or could there be some other issue causing the depthmap process stop to occur. I checked had the task log running in git bash and it displayed no issues. Screen capture below. 

Is there a photo processing restriction/limitation with the current software used in the ODM stack? 

Thanks! Please let me know if there is anything I could possibly do to help in resolving this. 

[image]

pierotofy, 2017-04-13:
Hi there! We don't impose any limit on the number of images, so as long as your system has enough power it should be able to process them.

With that said, this software is still beta, so things are not stable yet, and we've had some issues with the depthmap processing hanging.

If you could share the dataset that you are using (or a subset of it) so that a developer could try to reproduce the issue, that would be of tremendous help.

@dakotabenjamin have you seen this problem before?

mcjackattack, 2017-04-13:
I am getting this too. It will stop on a random image, process hangs on the depthmap, I tried updating and it does the same thing. I will post a link to the images I am using later, its only using 10% of one CPU(out of 10) so is not the processing power that is the issue.

jamesoutdoorrobotics, 2017-04-13:
@pierotofy Thanks for the reply! I have uploaded the same data set that I have been testing on Web and Node ODM. Here is the link for the set. 
[https://www.dropbox.com/s/emf6y4dciekyr7e/field%20test%20100%20feet%2080%20overlap.zip?dl=0]

pierotofy, 2017-04-17:
I haven't been able to test this yet, but I plan to in the next few days. Sorry for the delay.

jamesoutdoorrobotics, 2017-04-19:
@pierotofy No worries and thank you for taking a look into this!

pierotofy, 2017-04-27:
I ran the dataset a few times but it errored out because I didn't have enough disk space on the machine I was processing the dataset on. I'll retry again after cleaning up some space.

pierotofy, 2017-04-29:
Ok I confirm that this is getting stuck on the depthmap computation process. A task has been running for 2 days and it's stuck at ```Computing depthmap for image DJI_0138.JPG```.

I'll try a few things and report back my findings.

pierotofy, 2017-04-30:
I was able to get further by checking the `use-pmvs` option, which uses pmvs instead of opensfm to generate the point cloud. My process was then killed because I tested it on a machine with little RAM (4 GB), but I'm guessing it would have continued without issues from there.

@jamesoutdoorrobotics would you be able to try to reprocess the dataset with `use-pmvs`?

Also, it would be worthwhile to open an issue with https://github.com/mapillary/OpenSfM/tree/master/opensfm, since this seems a fault related to their code.

rakkenrol, 2017-05-01:
Dear All,

I had the same problem this weekend. After I set 'use-pmvs' then I was able to make a run. At the end I have got the following error message: _"refinePatchBFGS failed!"_
Do you have any idea to solve this?

Thank you!

dakotabenjamin, 2017-05-01:
The `refinePatchBFGS failed!` message is just a warning and can be safely ignored. You should get a successful run out of that.

rakkenrol, 2017-05-01:
Thanks @dakotabenjamin, I am sorry, but I am fogot to mentioned that at this point the process has stucked... :( So I am not able to download the results...

dakotabenjamin, 2017-05-01:
Can you post the error that causes it to fail? You may be running into memory issues.

rakkenrol, 2017-05-01:
Tehere is no error message. Is still running, but nothing happens. I allocated 12 GB RAM in Docker. Can I do anything else? Thanks a lot!

pierotofy, 2017-05-14:
I would open an issue with OpenSfM.  github.com/mapillary/OpenSfM/issues

pierotofy, 2017-10-19:
This issue was moved to http://community.opendronemap.org/t/processing-stops-at-depthmapping-no-issue-report/49

closed 2017-10-19 by pierotofy