WebODM/WebODM #150: Process exited with code 1

riddle911, 2017-04-14:
Hi, Mr Baldwin. After I installed the webodm, I come across this problem.
I uploaded serveral photos about 100 to the webodm. the result of processing is"Process exited with code 1".
I checked your wiki, you said it is because the cpu dose not match.
I'm not quite understand what you mean.
I run webodm on the cloud, which has 8CORE 32g Ram. I wonder is there anyting I need to talk with the vps hosting company?
thank you.

pierotofy, 2017-04-14:
Hello!

`Process exited with code 1` could mean a number of different things. Would you be able to copy/paste the output log for the task that failed? (When you expand the task details from the project list, there's a box where the full output of the process is reported)

riddle911, 2017-04-14:
@pierotofy 
here is the log.
Project 1
`[INFO]    Initializing OpenDroneMap app - Thu Apr 13 18:26:01  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   resize dir: /var/www/data/3afb5fd0-ea31-4d4d-b34e-7fa16d8be9bf/images_resize
[DEBUG]   Loading dataset from: /var/www/data/3afb5fd0-ea31-4d4d-b34e-7fa16d8be9bf/images_resize
[DEBUG]   /var/www/data/3afb5fd0-ea31-4d4d-b34e-7fa16d8be9bf/images_resize
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0415.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0522.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0539.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0452.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0567.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[DEBUG]   Loaded DJI_0443.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0411.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0527.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0428.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0570.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0412.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0461.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0466.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0504.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0434.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0537.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0489.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0555.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0545.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0420.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0516.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0401.JPG | camera: dji fc330 | dimensions: 2400 x 1800 | focal: 0.0 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[... cut ...]
/0/100/f        eth0         network        Virtio network device
/0/100/10       eth1         network        Virtio network device
/0/1            scsi1        storage        
/0/1/0.1.0      /dev/cdrom   disk           DVD reader
/1              veth3ffe999  network        Ethernet interface
/2              vethea5a366  network        Ethernet interface
/3              veth291ee6c  network        Ethernet interface
`

I tested 2 projects and looks like the errors are the same.
Thank for your help.

pierotofy, 2017-04-14:
That's a bug in OpenSfM (it was actually solved 3 days ago, but hasn't made it into OpenDroneMap yet): https://github.com/mapillary/OpenSfM/commit/24ee416c77b2e309e14ac3c1030474b08c01956c

The workaround at the moment is to make sure the EXIF tags of your images do not have fractions that divide by zero (for example, 5/0, 1/0, etc.).

Perhaps if you can .zip a few images from one of the datasets I could give you more precise directions.

pierotofy, 2017-04-14:
@smathermather is there a timeline for when the new changes to OpenSfM will be included in ODM?

riddle911, 2017-04-14:
ok, can you give me your email. I'll sent some images to you.

pierotofy, 2017-04-14:
Use this form: masseranolabs.com/contact/

pierotofy, 2017-04-14:
The exif tags on your images are missing the focal length information (it's currently set to zero).

To fix it, download exiftool and run the following from the directory where your images are stored:

```
exiftool -m -exif:FocalLength=5 *.JPG
```

Replace 5 with the value in millimeters for your camera.

riddle911, 2017-04-14:
yeah I  got it.
and another question, do you know the path of images which uploaded.
I searched all the machine but no result.

pierotofy, 2017-04-14:
They are stored inside a docker volume. If you are not familiar with docker I would recommend simply reuploading them.

I will close this issue for now.

closed 2017-04-14 by pierotofy

smathermather, 2017-04-15:
No timeline yet on OpenSfM and ODM, @pierotofy, but I recognize we need one. @Paulinus -- what's the timing in the gcp improvements?