WebODM/WebODM #151: Unable to run on SubDomain

CTemme, 2017-04-14:
on an ubuntu instance I have WebODM running on port 8000 and ngix running on port 80
My dns is configured as follows:
>'A' record: Hostname = "example.com" pointing to server IP addy
>'A' record: Hostname = "webodm.example.com" pointing to server IP addy
>CNAME: Hostname = "www.example.com" is alias of "example.com"

So all requests (example.com, webodm.example.com, www.example.com) goto ngix on port 80.
I have a ngix reverse proxy to forward subdomains to specific ports on same machines by this sites-enabled entry:
```
server{
  listen 80;
  server_name webodm.example.com;
  location /{
    proxy_pass http://localhost:8000;
  }
}
```
via: http://stackoverflow.com/questions/23649444/redirect-subdomain-to-port-nginx-flask

This seems to work since I'm able to navigate to the WebODM dashbord via webodm.example.com.
I am able to log in, view projects and complete admin tasks (create, update, delete users, projcts etc)

However,when trying to upload images to an existing project or successfully created new project I get the following error after a dozen images or a few dozen mbs are uploaded
>"Could not upload all files. An error occured. Please try again." 

(btw, typo: 'occurred' has 2 'r's  :)

However, by using  ip_addy:8000 to that same instance (rather than webodm.example.com) all my projects/tasks run to completion.

ubuntu 16.04 on a Docker Droplet 8gb ram

If its an issue with WebODM and is low priority, Id appreciate if anyone can suggest a workaround!

Thanks!
Chris

pierotofy, 2017-04-14:
Hi Chris, thanks for reporting the problem!

I think this is probably a configuration issue with Django, but I'm not 100% sure. If you could provide me with some credentials to log-in to the WebODM instance I could do some further troubleshooting (use this form if you need to contact me in private: masseranolabs.com/contact/). I'd rather not setup a subdomain myself just to narrow down this problem.

Thanks!

millerjs2, 2017-04-15:
For what it's worth, I'm running WebODM behind an Apache reverse proxy with no problems (I have also run it behind Caddy, but it's been a while since I used nginx). My guess is that something in your configuration is having a hard time with the images uploaded. If you only upload one or two images, does that work? Or do all uploads fail? Have you looked through error and access log for any clues?

Here's what my configuration looks like:
`<VirtualHost *:80>
        ProxyPreserveHost On
        ServerAdmin admin@mydomain.com
        ServerName webodm.mydomain.org
        ServerAlias webodm.mydomain.org
        ErrorLog ${APACHE_LOG_DIR}/error.log
        CustomLog ${APACHE_LOG_DIR}/access.log combined
        ProxyPass / http://webodmbox:8000/
        ProxyPassReverse / http://webodmbox:8000/
</VirtualHost>`

pierotofy, 2017-04-19:
@CTemme are you still having issues running WebODM on a subdomain?

pierotofy, 2017-04-27:
I'll close this for now until we receive more information.

closed 2017-04-27 by pierotofy