WebODM/WebODM #153: Failed native install Ubuntu 16.04.02 LTS

TheDude005, 2017-04-14:
During the npm install process I received this set of errors in the command line console:
npm WARN prefer global node-gyp@3.6.0 should be installed with -g

> node-sass@3.13.1 install /home/jordan/WebODM/node_modules/node-sass
> node scripts/install.js

sh: 1: node: not found
npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.1.1
npm ERR! Linux 4.4.0-71-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! node-sass@3.13.1 install: `node scripts/install.js`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the node-sass@3.13.1 install script 'node scripts/install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-sass package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node scripts/install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-sass
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-sass
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/jordan/WebODM/npm-debug.log

During the pip portion prior to this I got this error:
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-vRpUIW/psycopg2/

I have node.js and npm freshly installed prior to starting the WebODM install.

TheDude005, 2017-04-14:
Although I should have noted my GDAL would be inadequate for native install as the standard distro package has lagged so far behind the latest release of the bin files.

pierotofy, 2017-04-14:
Do you have both `nodejs` and `node` executables in your /usr/bin path?

`sudo ln -s `which nodejs` /usr/bin/node`

TheDude005, 2017-04-14:
I see nodejs when running dir in the /usr/bin directory but I don't see node

pierotofy, 2017-04-15:
Are you still having troubles installing WebODM? Did creating a symlink like I suggested above help your situation?

TheDude005, 2017-04-15:
I had removed the native install and used the Docker since my GDAL version was less than required, but I will do a native install again to see if the symlink let's the installation finish.

pierotofy, 2017-04-16:
I'll close this for now, if you still have questions post here! Thanks!

closed 2017-04-16 by pierotofy