WebODM/WebODM #155: Ability to share task datasets

pierotofy, 2017-04-14:
It would be convenient to allow users to share a task result (3D view, or map view, or both) with the public, without needing to provide credentials. Perhaps allowing anonymous users, or a one time link with a token.

More details of the use case:

```
I'd really like to be able to deploy it in a standalone fashion with a single dataset to make it easy to share models over the web
...
after I process the imagery, I can give them a tiff, and that's great, but the ply/las files are a bit harder for them to visualize on their own
and the web viewer just works so well and I can leave it hosted on my servers
the ability to link straight through to a 3D viewer for a given task would be really handy
```

smathermather, 2017-04-15:
This is a great idea. I like the idea of a token.

pierotofy, 2017-12-01:
Work in progress: 

[image]

pierotofy, 2017-12-05:
Done! For an example: https://demo.webodm.org/map/project/3/task/904c1c56-10c5-48df-a104-0138f699dd69/

closed 2017-12-05 by pierotofy