WebODM/WebODM #157: Task failing with error (no such file or directory)

pmumby, 2017-04-18:
I've tried uploading a test mapping job I ran using my Mavic Pro, and DroneDeploy app.
Map was captured using 42 Photos, from 200 feet altitude, with 80% sidelap and 80% frontlap (so should have 80% overlap in both directions).

I'm using fresh installation of latest version of WebODM, new job, uploaded photos, and used default settings everywhere else.

The same process using the Aukerman test data worked just fine. So not sure what's wrong with this data. I'll begin debugging/trying to find the issue myself, but wanted to post for help as well, in case it's something easy to fix.

WebODM is running in default config (on docker) on an Ubuntu 16.04 Virtual Machine, which is configured for 4 vCPUs, and 24GB of RAM. The host is a Linux VM Host (QEMU/KVM), which has a quad physical core xeon, (8 logical cores). and approx 200GB of hard disk.

The Aukerman dataset finishes in about 40min, but my test data crashes with error after approx 3-4 minutes.

Here are the full logs from the run, with errors (this is only a test dataset, so nothing sensitive) I'll also gladly attach the actual image data if that would help. Let me know.
```
[INFO]    Initializing OpenDroneMap app - Tue Apr 18 15:02:08  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/34f3d38f-ae3e-4307-b9e4-50db8b3a7425/images
[DEBUG]   /var/www/data/34f3d38f-ae3e-4307-b9e4-50db8b3a7425/images
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0015.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0043.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0005.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0019.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0003.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0020.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0032.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0014.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0041.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0037.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0004.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0021.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0042.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0007.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0027.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0006.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0018.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0028.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0039.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0035.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0031.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0034.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0025.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0030.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0008.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0038.JPG | camera: dji fc220 | dimensions: 4000 x 3000 | focal: 4.73 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[... 573 lines cut ...]
2017-04-18 15:05:15,127 Full matching 30 / 31, time: 1.48131895065s
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm create_tracks /var/www/data/34f3d38f-ae3e-4307-b9e4-50db8b3a7425/opensfm
2017-04-18 15:05:15,816 reading features
2017-04-18 15:05:17,224 Merging features onto tracks
2017-04-18 15:05:17,318 Good tracks: 4392
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm reconstruct /var/www/data/34f3d38f-ae3e-4307-b9e4-50db8b3a7425/opensfm
2017-04-18 15:05:18,643 0 partial reconstructions in total.
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm mesh /var/www/data/34f3d38f-ae3e-4307-b9e4-50db8b3a7425/opensfm
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
[... cut ...]
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/var/www/data/34f3d38f-ae3e-4307-b9e4-50db8b3a7425/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 106, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

pierotofy, 2017-04-18:
Hi, thanks for reporting the issue!

It looks like OpenSfM wasn't able to generate a reconstruction from the input images. Would you be able to share your dataset either publicly or privately for us to try to process it?

I would try to increase the `min-num-features` parameter. As an alternative, you could also try to check the `use-pmvs` which will use PMVS instead of OpenSfM for the point cloud reconstruction.

smathermather, 2017-04-18:
With a Mavic Pro, is that an wide angle camera? If so, I would recommend calibrating the images before inputting them into ODM.

pmumby, 2017-04-18:
Mavic is wide angle, not as wide as Phantom for example, but still "wide". I believe it's a 28mm equivalent.

Can you elaborate on the "calibration" process? Is this something we can do to save for the mavic camera in some metadata file somewhere within ODM? Or is this a process that needs to be done on every image?

I'll take a look at those options, and I'll gladly share the dataset, 
http://www.orb0.com/files/ShirleysBay.zip

Let me know if that zip works for you, if not I'll share via another method of your preference.

pmumby, 2017-04-18:
Also quick update, I'm open to the idea I just screwed up the capture too :)

This was my first attempt at mapping, and I just used DroneDeploy as I thought their app seemed decent for controlling the flight plan on the mavic.

I've got a trip on Saturday (the 22nd) where I'd like to capture mapping data, and wanted to do a dry-run to figure out how it works. On Saturday I need to do a 15acre mapping exercise, and I won't be going back for at least a month, so I need to at least capture the data correctly while there. 

I'll possibly have an opportunity before that for another trial run, but once you've looked at the dataset, any other tips you can offer (or if you're familiar with any better app that works on the DJI drones on Android) would be greatly appreciated.

smathermather, 2017-04-19:
24mm effective isn't too bad, and typically doesn't require pre-calibration, so it's possible you didn't fly with adequate overlap. I haven't tried the DroneDeploy app in a while, but do you recall if you set a percent overlap for your flight?

pmumby, 2017-04-19:
I set the overlap to 80% both "frontlap" and "sidelap". However it appears the app didn't do this correctly.

Looking at the images in the set, there appears to be almost zero overlap between the images.

I wonder if the app is expecting a wider angle lens like on some phantom models, and did the route planning incorrectly as a result? (and shutter timing)

Anyone have any suggestions for another app that works on Android for DJI drones? I've seen 2 others that seem popular, Pix4d, and PrecisionFlight both seem half decent. Pix4d bothers me a bit (seems a bit sketchy the way it requires a separate app to control the drone, which seems half-baked). But PrecisionFlight actually looks pretty good.

Unfortunately hard to tell until I actually take one of these out to the field.

Not sure if I'll have time to test before I take my trip, but I'll try.

Any suggestions would be appreciated.

At least it looks like the problem is definitely the data-set, probably the lack of overlap. In the short term I can always set the overlap to 100% and see what happens. (it won't let me go higher than that, and 80% resulted in zero overlap in the images). Very odd that it resulted in near zero overlap, that's a HUGE discrepancy from the expected 80% I set in the settings for the flight.

millerjs2, 2017-04-19:
The images themselves look pretty good (no barrel distortion/etc), but it looks like you could use a lot more overlap. As for apps, I use Pix4D Capture with my 3DR Solo with great results - I'd test a few apps and see how they do. 

Also, for reference, here's what your points & flightpath look like:
[image]

The blue dots are the pictures you took, the red dots are pictures I believe should have also been taken. You have a decent amount of overlap to either side, but not enough facing forward. 

For proof, look at this output from Pix4D:
[image]

That entire polygon should be dark green (5+). You flew an area of about 160m x 215m and only got 42 images out of it. Even at 200ft altitude I'd expect to get about 3-4 the amount of pictures you have here. 

Good luck, and happy testing!

pierotofy, 2017-04-19:
Notice also that the area flown has a lot of vegetation, so definitely more overlap would help, but also increase the `min-num-features` parameter to something like `12000`.

pierotofy, 2017-05-04:
I'll close this for now, unless there are further questions.

closed 2017-05-04 by pierotofy