WebODM/WebODM #158: Any way to improve the orthoimage/3d framework

riddle911, 2017-04-19:
I have succeeded in aerial map processing.
There are 2 more problems
1.The orthoimage is kind of broken...I mean it always lost something.
2.the 3d vision funciton is nice, but the quality of the model looks not so good.
I found there is a JS framework named **Cesium** , I wonder if it could be updated...
Appreciate your work !

pierotofy, 2017-04-19:
Congratulations on the map processing! 🎉 

The fact that the orthophoto and 3D model look broken suggests that there might be some problems with either the images you are using, or some parameters might need tweaking.

If you could share a link to the dataset here we might be able to offer suggestions on how to improve the results.

Adding Cesium will not increase the quality of the models (but it's a great framework!).

riddle911, 2017-04-20:
Thank for your help!
I'm using the test data _odm_data_aukerman-master_,here is the [orthophoto](http://7xvmzt.com1.z0.glb.clouddn.com/odm_data_aukerman-master/odm_orthophoto/odm_orthophoto.png) you may download.
I used the default parameters..

dakotabenjamin, 2017-04-20:
I looked at the orthophoto and it looks quite good. There are some holes around the edges and where we didn't get good enough overlap for the trees. This is normal and an artefact of the image overlap. I would suggest increasing the mesh-size, the mesh-octree-depth, and orthophoto-resolution to improve the overall quality of the 3D textured mesh and orthophoto.

riddle911, 2017-04-21:
Thank for your help. pierotofy and dakotabenjamin 
It's very kind of you.

closed 2017-04-21 by riddle911