WebODM/WebODM #165: Options display not scrollable and as a result changed options can't be saved

simonpoole, 2017-04-29:
Running WebODM from the repo of today.

Run a task and then: Edit task -> Set Options 

The resulting list of options is (very very) long and does not fit on the screen and at the same time isn't scrollable, with the result the that save button is virutally something like a yard below my desk .

Observed with both current FF and Chrome versions on W10

pierotofy, 2017-04-29:
Hi @simonpoole thanks for reporting the problem.

I confirm this is an issue. I've been able to scroll using the mouse wheel, but I realize that not everyone uses a mouse nowadays.

pierotofy, 2017-05-01:
This is now fixed with the latest master.

[image]

closed 2017-05-01 by pierotofy