WebODM/WebODM #166: Native Install Errors

macsee, 2017-04-30:
Hi! I decided to do a native installation on Ubuntu 17.04 but got some errors. I followed the steps published in this repo for a native install but I got some postgres errors: FATAL:  password authentication failed for user "postgres".
I've install Postgres (9.6) and created the local_settings.py file in the WebODM/webodm folder, as sugested, but nothing happens. Also I can't figure it out how to connect to the database to set postgis.enable_outdb_rasters and postgis.gdal_enabled_drivers settings. Using pgadmin I can't see the database. At what moment during the installation steps this database is created? Or do I have to create it manually?
A little help will be appreciated.
Cheers!

pierotofy, 2017-04-30:
Hi @macsee, you need to create the database manually. Then you need to configure local_settings.py with the proper parameters (username, password, database name).

macsee, 2017-04-30:
Ok. I'll do it. Thanks @pierotofy!

pierotofy, 2017-04-30:
Thank you for pointing out the missing step! I've clarified this step in the readme.

closed 2017-04-30 by pierotofy

macsee, 2017-07-13:
Hello @pierotofy ,
Sorry to open this again. I wasn't sure if I had to open a new topic or use this one, since the problem is similar.
I wanted to install WebODM on a P.C with a fresh installation of Ubuntu 16.04. I followed the steps for native install and I've got the following output when executing `start.sh`:

Checking python version... 3.x, good!
Checking GDAL version... GDAL 2.1.3, released 2017/20/01, excellent!
Running migrations
INFO Booting...
WARNING Debug mode is ON (for development this is OK)
INFO Starting background scheduler...
Operations to perform:
  Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
Running migrations:
  No migrations to apply.
WARN: /webodm/build/static does not exist, CSS, JS and other files might not be available.
nginx: [alert] could not open error log file: open() "/var/log/nginx/error.log" failed (13: Permission denied)
2017/07/12 22:34:35 [warn] 17883#17883: the "user" directive makes sense only if the master process runs with super-user privileges, ignored in /home/macsee/WebODM/nginx/nginx.conf:4
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] bind() to 0.0.0.0:8000 failed (98: Address already in use)
nginx: [emerg] still could not bind()
[2017-07-12 22:34:38 -0300] [17884] [INFO] Starting gunicorn 19.7.1
[2017-07-12 22:34:38 -0300] [17884] [INFO] Listening at: unix:/tmp/gunicorn.sock (17884)
[2017-07-12 22:34:38 -0300] [17884] [INFO] Using worker: sync
[2017-07-12 22:34:38 -0300] [17890] [INFO] Booting worker with pid: 17890

Congratulations! └@(･◡･)@┐
==========================

If there are no errors, WebODM should be up and running!

Open a web browser and navigate to http://localhost:8000

When I open a web browser, everything looks as if CSS and Javascript files are missing. What's the problem now? I have also tried executing this script: `contrib/ubuntu_1604_install.sh` but nothing different happens. I dont know what else to try. Can you help me please? 
Thanks