WebODM/WebODM #167: old docker images being kept indefinitely

ctrl6DB, 2017-05-05:
It seems that when running webodm.sh update, it builds a new image for webodm_webapp , but it keeps previous images.
Considering that the image size is in excess of 12GB (on Ubuntu, native docker engine), is it possible to remove previous images after a successful build ?
I use `docker rmi <IMAGE>` , but it would nice to have the option to automatically remove old builds.

Thank you.

pierotofy, 2017-05-05:
Hi @ctrl6DB, yes this is a good suggestion, old images should be removed.

Thanks for pointing it out!

pierotofy, 2017-05-06:
This should be now fixed: https://github.com/OpenDroneMap/WebODM/commit/83e176e70039041b7baf5057258e49f8b25a0fe0

closed 2017-05-06 by pierotofy

ctrl6DB, 2017-05-12:
Thank you very much ! Indeed, it works great.