WebODM/WebODM #168: Bad functioning of Web UI on mobiles

macsee, 2017-05-06:
Hi!
I successfully connected to the Web UI of WebODM, which runs on my PC, using my mobile's browser (Chrome for iPhone 6S in this case), but when I tap on my first task, the top one, nothing happens. I can't see console output or view map. But here is some strange behavior. When I tap on any task below the first one, they work! 
I think the issue appears when the Bootstrap navbar is shown. Checking with the Chrome developer tools I can see that It's overlapping the click area of the first task, so that is why isn't working. It is a little bug, is not really important, but I wanted to share whith you guys.
Cheers!

pierotofy, 2017-05-06:
Hi @macsee, thanks for reporting the problem!

We have not started on adding support for mobile, so I would expect UI problems to be present (I'm surprised nobody pointed it out sooner :) ).

I'll leave this issue open until we have the UI fixed for mobile support.

macsee, 2017-05-06:
Ok @pierotofy. In the meantime I'll try to fix it locally. 
Thanks!

pierotofy, 2017-05-07:
Awesome! Follow these guidelines to edit the various CSS files: https://opendronemap.github.io/WebODM/#setup-a-development-environment

If you are able to improve the display, would you consider opening a pull request and contribute the changes you make? Even small improvements count 👍

pierotofy, 2017-05-07:
Also keep an eye on https://github.com/gkconsulting/WebODM/commit/ed8b89ecea39b91c53682554d1762cf71e9d18ba where it looks like they have started adding some mobile improvements.

macsee, 2017-05-07:
@pierotofy Definitely!

macsee, 2017-05-09:
UPDATE: Apparently this was solved on gkconsulting@ed8b89e

pierotofy, 2017-05-14:
I've ported the changes from gkconsulting@ed8b89e into master.

closed 2017-05-14 by pierotofy