WebODM/WebODM #169: Question: Temp data storage

AnthonyCoughlin, 2017-05-11:
When uploading images to webodm, are the images actually uploaded and stored in a webodm folder somewhere on my machine or does webodm just use the actual location of the images?  If they are uploaded and stored in a second location are they purged/deleted upon completion of a project or do I need to delete them manually?  If manual deletion is necessary, could you please help me locate them and delete them!  Thanks in advance.

Reason for the question, I'm seeing some high storage totals but am unable to actually locate the source.  Thought it might be odm...

Anthony

pierotofy, 2017-05-11:
Hi @AnthonyCoughlin, the images are stored on a docker volume, and they are not removed after the task is completed. If you want to free up space, I recommend that you delete a task after you have downloaded its assets.

The node-opendronemap docker volume will also keep a copy of the images for 3 days. So the total space taken is 3 times the size of your dataset.

AnthonyCoughlin, 2017-05-11:
awesome, thanks.  any chance you can provide me with a line of code to run in docker command line to see how much space is being used in the docker volume?