WebODM/WebODM #172: Running Out of Memory

Pratyush1991, 2017-05-18:
While trying to process a dataset of 174 images taken by iphone5s but it runs out of memory. At first I tried processing by assigning 12GB of RAM and with 6 cores and then with 14GB RAM and 7 cores. In both the cases the process ran out of memory which I feel should be enough to process 174 images. So I'm not sure what is the problem. The error log as well as link to my data set have been attached below.
I will be glad if you could help..:-)
  
https://drive.google.com/drive/folders/0B7tV_P5gZivFYkJPTUtieTBJNzg?usp=sharing

[image]

	Blending texture patches 100%... done. (Took 20.075s)
Generating texture atlases:
	Sorting texture patches... done.
done.

	Working on atlas 41 100%... done.
	Finalizing texture atlases... done. (Took: 3.698s)
Building objmodel:
	Saving model... 	Saving model... 	Saving model... done.
Whole texturing procedure took: 88.492s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/dd588376-1fd5-4158-b3d2-961dc842e9f3/odm_georeferencing/coords.txt
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.

[WARNING] Georeferencing failed. Make sure your photos have geotags in the EXIF or you have provided a GCP file. 
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM Orthophoto Cell
[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /var/www/data/dd588376-1fd5-4158-b3d2-961dc842e9f3/odm_texturing/odm_textured_model.obj -logFile /var/www/data/dd588376-1fd5-4158-b3d2-961dc842e9f3/odm_orthophoto/odm_orthophoto_log.txt -outputFile /var/www/data/dd588376-1fd5-4158-b3d2-961dc842e9f3/odm_orthophoto/odm_orthophoto.png -resolution 20.0  -outputCornerFile /var/www/data/dd588376-1fd5-4158-b3d2-961dc842e9f3/odm_orthophoto/odm_orthophoto_corners.txt
    plasm.execute(niter=1)
  File "/code/scripts/odm_orthophoto.py", line 67, in process
    '-outputCornerFile {corners}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

pierotofy, 2017-05-18:
Long story short, being that the images are not georeferenced, the model is without a reference scale and the orthophoto tries to generate a really large image (which does not fit in memory).

I was able to process the dataset by lowering the --resolution parameter to something around 4 (down from 20).

[image]

I've also opened a PR to return a more meaningful error message next time this happens.

closed 2017-05-18 by pierotofy

Pratyush1991, 2017-05-18:
Thanks a lot..!! 
I also tried by reducing these photos to around 100 which worked fine. I would be happy if you could suggest how can  I improve  the texture.

pierotofy, 2017-05-18:
For the orthophoto, try turning on the `--use-25dmesh` option.

For the textured 3D model, try increasing the `--mesh-octree-depth` and `--mesh-size` parameters.