WebODM/WebODM #176: Keep getting no space on Device have plenty of HD space but VM machine wont seem to expand past 19 gig

marvinlange, 2017-05-21:
`Environment:

Request Method: GET
Request URL: http://192.168.99.102:8000/processingnode/1/

Django Version: 1.10.5
Python Version: 3.5.3
Installed Applications:
['django.contrib.admin',
 'django.contrib.auth',
 'django.contrib.contenttypes',
 'django.contrib.sessions',
 'django.contrib.messages',
 'django.contrib.staticfiles',
 'django.contrib.gis',
 'guardian',
 'rest_framework',
 'rest_framework_nested',
 'webpack_loader',
 'corsheaders',
 'app',
 'nodeodm']
Installed Middleware:
['corsheaders.middleware.CorsMiddleware',
 'django.middleware.security.SecurityMiddleware',
 'django.contrib.sessions.middleware.SessionMiddleware',
 'django.middleware.common.CommonMiddleware',
 'django.middleware.csrf.CsrfViewMiddleware',
 'django.contrib.auth.middleware.AuthenticationMiddleware',
 'django.contrib.messages.middleware.MessageMiddleware',
 'django.middleware.clickjacking.XFrameOptionsMiddleware',
 'django.middleware.locale.LocaleMiddleware']

Traceback:

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  64.                 return self.cursor.execute(sql, params)

The above exception (could not extend file "base/16384/18022": No space left on device
HINT:  Check free disk space.
) was the direct cause of the following exception:

File "/usr/local/lib/python3.5/site-packages/django/core/handlers/exception.py" in inner
  39.             response = get_response(request)

File "/usr/local/lib/python3.5/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)

File "/usr/local/lib/python3.5/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/usr/local/lib/python3.5/site-packages/django/contrib/auth/decorators.py" in _wrapped_view
  23.                 return view_func(request, *args, **kwargs)

File "/webodm/app/views.py" in processing_node
  86.     if not pn.update_node_info():

File "/webodm/nodeodm/models.py" in wrapper
  26.             return func(*args, **kwargs)

File "/webodm/nodeodm/models.py" in update_node_info
  78.             self.save()

File "/usr/local/lib/python3.5/site-packages/django/db/models/base.py" in save
  796.                        force_update=force_update, update_fields=update_fields)

File "/usr/local/lib/python3.5/site-packages/django/db/models/base.py" in save_base
  824.             updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)

File "/usr/local/lib/python3.5/site-packages/django/db/models/base.py" in _save_table
  889.                                       forced_update)

File "/usr/local/lib/python3.5/site-packages/django/db/models/base.py" in _do_update
  939.         return filtered._update(values) > 0

File "/usr/local/lib/python3.5/site-packages/django/db/models/query.py" in _update
  654.         return query.get_compiler(self.db).execute_sql(CURSOR)

File "/usr/local/lib/python3.5/site-packages/django/db/models/sql/compiler.py" in execute_sql
  1148.         cursor = super(SQLUpdateCompiler, self).execute_sql(result_type)

File "/usr/local/lib/python3.5/site-packages/django/db/models/sql/compiler.py" in execute_sql
  835.             cursor.execute(sql, params)

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  79.             return super(CursorDebugWrapper, self).execute(sql, params)

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  64.                 return self.cursor.execute(sql, params)

File "/usr/local/lib/python3.5/site-packages/django/db/utils.py" in __exit__
  94.                 six.reraise(dj_exc_type, dj_exc_value, traceback)

File "/usr/local/lib/python3.5/site-packages/django/utils/six.py" in reraise
  685.             raise value.with_traceback(tb)

File "/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py" in execute
  64.                 return self.cursor.execute(sql, params)

Exception Type: OperationalError at /processingnode/1/
Exception Value: could not extend file "base/16384/18022": No space left on device
HINT:  Check free disk space.
`

marvinlange, 2017-05-21:
OperationalError at /processingnode/1/

could not extend file "base/16384/18022": No space left on device
HINT:  Check free disk space.

Request Method: 	GET
Request URL: 	http://192.168.99.102:8000/processingnode/1/
Django Version: 	1.10.5
Exception Type: 	OperationalError
Exception Value: 	

could not extend file "base/16384/18022": No space left on device
HINT:  Check free disk space.

Exception Location: 	/usr/local/lib/python3.5/site-packages/django/db/backends/utils.py in execute, line 64
Python Executable: 	/usr/local/bin/python
Python Version: 	3.5.3
Python Path: 	

['/webodm',
 '/webodm',
 '/usr/local/lib/python35.zip',
 '/usr/local/lib/python3.5',
 '/usr/local/lib/python3.5/plat-linux',
 '/usr/local/lib/python3.5/lib-dynload',
 '/usr/local/lib/python3.5/site-packages']

Server time: 	Sun, 21 May 2017 21:29:51 +0000

pierotofy, 2017-05-22:
Hi @marvinlange I'm assuming you are on Windows.

Make sure you allocate enough disk space to your Docker container. By default it will stop at 20G. See this link https://github.com/docker/kitematic/issues/825

dani2007, 2017-05-22:
This Video explains it very vell: https://youtu.be/kJQegRQUhWU?t=5m54s

He links to: http://support.divio.com/local-development/docker/managing-disk-space-in-your-docker-vm

you basicly add a Flag at the right place in the "start.sh" from docker. So Docker will creat a bigger diks in the Virtual machine. 

But at the same Time i have a question myself:
I would like to move the virtual machine to my secondary hard disk, since my first one ist too small. Does somebody know how I could do this?

pierotofy, 2017-05-22:
Nice link @dani2007 !

I think I will add that to both the README and within WebODM.

pierotofy, 2017-05-22:
@dani2007 you need to move the virtual disk to your secondary hard drive. See this link: https://dirkstrauss.com/moving-virtualbox-vm/

marvinlange, 2017-05-22:
I upgraded to windows 10 and now its using a file MobyLinuxVM.vhdx  at 48 GIG seems to be processing fine but the view 3d Model dos not work 
[image]

marvinlange, 2017-05-22:
Found it sorry it was a Firefox issue it works in Microsoft Edge

closed 2017-05-22 by pierotofy