WebODM/WebODM #181: Cant upload images

marvinlange, 2017-05-25:
There are no usable processing nodes. We tried to reach:

    node-odm-1:3000 (queue: 0)

Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.

things I have tried
1 Added all permissions to admin
2 Reinstalled webodm many times

I have spent days trying to resolve this any help will be greatly appreciated 

[docker_logFile.txt](https://github.com/OpenDroneMap/WebODM/files/1029678/docker_logFile.txt)
[webodm-log.txt](https://github.com/OpenDroneMap/WebODM/files/1029680/webodm-log.txt)

marvinlange, 2017-05-25:
I checked box keep stdin open in kitematic and started node
Update I deleted the Node and recreated it using web page it now shows processing but it is not sending log data to the web page thru api  seems like I have some sort of api communication issue

pierotofy, 2017-05-25:
I think you might have something installed on your computer that's preventing docker's network bridge from working properly. Perhaps try to uninstall or disable `vpnkit.exe` and try again?

marvinlange, 2017-05-25:
Doing hail marry system restore as it was working :)

marvinlange, 2017-05-25:
pierotofy  you were Correct !!! I but I had changed so much trying to fix the BSOD i was getting from windows 10 after enabling Hyper -V I was not sure what I may have done wrong so I did system restore and at least webodm is back up. now i will get back to trying to solve BSOD ..

ALSO I do now see 2 Hyper V network controllers were I had only seen one wend I was having Issue :)

Thanks Again
GOD Bless

marvinlange, 2017-05-25:
UGG now its giving me different error

` <tempfile._TemporaryFileCloser object at 0x7f36a55c7b38>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpv7ar_yvi.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f36a55c7d30>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmp5__1ur9u.upload'
webapp        | [2017-05-25 23:43:27 +0000] [439] [INFO] Booting worker with pid: 439`

closed 2017-05-27 by marvinlange