WebODM/WebODM #182: Process exited with code 1

dstrohsch, 2017-05-26:
I used Webodm with photos from a Mapir Survey2 camera and the images seemed to be processed correctly. However, when I try processing images from the Mapir Survey1 camera I get "Process exited with code 1". I am running on an 8GB RAM Windows 7 Professional laptop.

I have uploaded the seven sample images to this dropbox URL: https://www.dropbox.com/sh/tkh8yltpnqx84wz/AACunJ9u-jMY4lEDBP_QZGNCa?dl=0

Here is the output from the task window:

[INFO]    Initializing OpenDroneMap app - Thu May 25 23:56:46  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /var/www/data/796581de-e80b-45be-83df-1eb687e92af7/images
[DEBUG]   /var/www/data/796581de-e80b-45be-83df-1eb687e92af7/images
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092855_029.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092848_027.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092851_028.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092901_031.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092907_033.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092904_032.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded 2017_0520_092858_030.JPG | camera: peau productions mapir | dimensions: 4032 x 3024 | focal: 4.35 | ccd: None
[INFO]    Found 7 usable images
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM Resize Cell
[DEBUG]   Resizing dataset to: /var/www/data/796581de-e80b-45be-83df-1eb687e92af7/images_resize
[DEBUG]   Resized 2017_0520_092855_029.JPG | dimensions: (1800, 2400, 3)
[DEBUG]   Resized 2017_0520_092848_027.JPG | dimensions: (1800, 2400, 3)
[DEBUG]   Resized 2017_0520_092851_028.JPG | dimensions: (1800, 2400, 3)
[DEBUG]   Resized 2017_0520_092901_031.JPG | dimensions: (1800, 2400, 3)
[DEBUG]   Resized 2017_0520_092907_033.JPG | dimensions: (1800, 2400, 3)
[DEBUG]   Resized 2017_0520_092904_032.JPG | dimensions: (1800, 2400, 3)
[DEBUG]   Resized 2017_0520_092858_030.JPG | dimensions: (1800, 2400, 3)
[INFO]    Resized 7 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm extract_metadata /var/www/data/796581de-e80b-45be-83df-1eb687e92af7/opensfm
2017-05-25 23:56:49,645 Extracting focal lengths for image 2017_0520_092855_029.JPG
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/extract_metadata.py", line 29, in run
    d = exif.extract_exif_from_file(data.load_image(image))
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 83, in extract_exif_from_file
    d = exif_data.extract_exif()
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 247, in extract_exif
    focal_35, focal_ratio = self.extract_focal()
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 165, in extract_focal
    get_float_tag(self.tags, 'EXIF FocalLengthIn35mmFilm'),
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 27, in get_float_tag
    return float(tags[key].values[0])
AttributeError: Ratio instance has no attribute '__float__'
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 85, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

Thanks in advance for the help.
-Dave

pierotofy, 2017-05-26:
Hi Dave,

It looks like your camera is setting the "FocalLengthIn33mmFilm" EXIF tag in a non-standard manner. Instead of being defined as a Short datatype (http://www.exiv2.org/tags.html) it's defined as a Rational:

```
import exifread
exifread.process_file(open('/datasets/2017_0520_092851_028.JPG'), details=False)
# ...
(0xA405) Ratio=31 @ 846
```

You can fix your tags by running exiftool:

```
for f in /path/to/your/images/*.JPG; do exiftool -all= -tagsfromfile @ -all:all -unsafe -icc_profile $f; done
```

http://www.sno.phy.queensu.ca/~phil/exiftool/

It should work properly after that.

pierotofy, 2017-05-26:
@dakotabenjamin have you seen other people reporting this type of problem? I wonder if we could just sanitize the images by running them through the exiftool as part of the processing chain.

dakotabenjamin, 2017-05-26:
A while back I fixed a tagging issue but I don't remember exactly what the problem was. I agree with this solution. Is this an issue with pyexiv2?

pierotofy, 2017-05-26:
I think it's just violation of the exif standard by camera vendors.

dakotabenjamin, 2017-05-26:
OK, then sanitization is the best option.

dstrohsch, 2017-05-26:
Piero and Dakota,
  Thanks for the suggestions and the code! I implemented it in a .bat file
(I'm on Windows) and then processed the pictures. All results were
generated correctly and without any errors being generated.
So, I'm considering this issue closed from my side.
Also, a big thanks for responding so quickly!
-Dave

pierotofy, 2020-04-02:
We should have much better support for EXIF tags in newer versions of ODM. If this is still a problem, re-open this issue? :clinking_glasses:

closed 2020-04-02 by pierotofy