WebODM/WebODM #183: Question on installing Narive in Winows

marvinlange, 2017-05-27:
I am beyond frustrated with Docker and windows I even upgraded my windows 7 tat I was very happy with to windows 10 thinking that it would help :(
So
I decided to run it natively its now a day later and I have a couple of hairs left after going thru prerequisites.

But for the life of me I cannot figure out how to install Apache + mod_wsgi ......  :(

I am stuck in the list at the foiling line on native install
 Windows users should refer to this guide to install Apache + mod_wsgi and run gunicorn:

I do already have XAMP installed can I just use that ?
I just want a Stable Version of webodm and docker kept blowing up on me maybe the anser is to take a spare PC I have and try doing a Linux Ubuntu webodm server.

Once running I will even share it as a Test/Demo server to anyone who needs it as I do have good hot 5meg up 10meg down Internet  I would even be wiling to do a fresh install and open up a port and let someone remote in to build a dedicated Web-ODM Machine and then they can use it anytime I am just beyond frustrated, Mostly with how broken Docker is and my lack of Linux knowledge 

Thanks in advance !!!
 I hate asking for to much help but I have spent almost 3 weeks almost everyday trying to just get one stable Web-Odm system working.... 
My wife is mad at me and I need my life Back :(

GOD Bless
Marvin
marvinlange(at)gmail.com

marvinlange, 2017-05-27:
Before Microsoft Bought out this little company that made Virtual PC it was the Greatest little Software now is integrated into windows and Blown up :(

marvinlange, 2017-05-27:
1. - Well Finally got Apache installed and verified running
2..  SQL Data Base Working 
3. - now any advice on getting mod_wsgi 4.5.15 to install

>  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\Apache24/include -Ic:\users\marvin\appdata\local\programs\python\python36\include -Ic:\users\marvin\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\winrt" /Tcsrc/server\
 mod_wsgi.c
 c:\users\marvin\appdata\local\programs\python\python36\include\pyconfig.h(307): warning C4005: 'PLATFORM': macro redefinition
 c:\apache24\include\os.h(42): note: see previous definition of 'PLATFORM'
 src/server\mod_wsgi.c(469): warning C4267: 'initializing': conversion from 'size_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(470): warning C4267: 'initializing': conversion from 'size_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(578): warning C4267: '=': conversion from 'size_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(727): warning C4267: '=': conversion from 'size_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(773): warning C4267: '=': conversion from 'size_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(1858): warning C4244: '=': conversion from 'Py_ssize_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(3387): warning C4244: '=': conversion from 'Py_ssize_t' to 'long', possible loss of data
 src/server\mod_wsgi.c(5892): warning C4244: 'return': conversion from '__int64' to 'long', possible loss of data
 C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\Apache24/include -Ic:\users\marvin\appdata\local\programs\python\python36\include -Ic:\users\marvin\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\winrt" /Tcsrc/server\
 wsgi_apache.c
 c:\users\marvin\appdata\local\programs\python\python36\include\pyconfig.h(307): warning C4005: 'PLATFORM': macro redefinition
 c:\apache24\include\os.h(42): note: see previous definition of 'PLATFORM'
 C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\Apache24/include -Ic:\users\marvin\appdata\local\programs\python\python36\include -Ic:\users\marvin\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\winrt" /Tcsrc/server\
 wsgi_buckets.c
 c:\apache24\include\os.h(42): warning C4005: 'PLATFORM': macro redefinition
 c:\users\marvin\appdata\local\programs\python\python36\include\pyconfig.h(307): note: see previous definition of 'PLATFORM'
 C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\Apache24/include -Ic:\users\marvin\appdata\local\programs\python\python36\include -Ic:\users\marvin\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\winrt" /Tcsrc/server\
 wsgi_convert.c
 src/server\wsgi_convert.c(93): warning C4244: '=': conversion from 'Py_ssize_t' to 'long', possible loss of data
 C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Ic:\Apache24/include -Ic:\users\marvin\appdata\local\programs\python\python36\include -Ic:\users\marvin\appdata\local\programs\python\python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.15063.0\winrt" /Tcsrc/server\
 wsgi_daemon.c
 c:\apache24\include\os.h(42): warning C4005: 'PLATFORM': macro redefinition
 c:\users\marvin\appdata\local\programs\python\python36\include\pyconfig.h(307): note: see previous definition of 'PLATFORM'
[... cut ...]
 mod_wsgi.obj : warning LNK4197: export 'PyInit_mod_wsgi' specified multiple times; using first specification
    Creating library build\temp.win-amd64-3.6\Release\src/server\mod_wsgi.cp36-win_amd64.lib and object build\temp.win-amd64-3.6\Release\src/server\mod_wsgi.cp36-win_amd64.exp
 Generating code
 Finished generating code
 LINK : fatal error LNK1158: cannot run 'rc.exe'
 error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\link.exe' failed with exit status 1158

pierotofy, 2017-05-29:
@marvinlange note that you will always need docker on Windows to process a task; WebODM can run natively, but node-OpenDroneMap/OpenDroneMap still needs docker to operate.

https://github.com/OpenDroneMap/WebODM#opendronemap-node-opendronemap-webodm-what

With this information, are you still looking to run WebODM natively?

marvinlange, 2017-05-30:
YES in Windows and Linux as nodes 

I cannot seem to get it to run for any length of time 
I have found and Fixed Blue Screen issue : It seems that if I use Hyper -V Manager instead of Docker to add resources Ram,HD,CPU Cores I no longer Blue Screen.
I have had no luck installing docker natively on Linux Ubuntu 16.04: I believe the Scrip makes download request to unavailable software. the following would not download for me
`sudo apt-get install -y postgresql-9.6
sudo apt-get install -y postgresql-9.6-postgis-2.3`
so I ended up installing manually 
--got docker installed 
Can get web-odm to start  can see web server on Linux box from my PC.  but cant seem to find or connect to web odm page.

So I am back to using Docker on my PC seems to be stable but now for some reason it keeps complaining about cant upload images if I do over like 50 { 50gig hd space allocated, 12 gigs ram allocated, with 4 CPU cores )
and if they do upload then it complains it cant find processioning node :(
the sad part I have done some huge projects 800 + photos was working great until I shut down or rebooted host PC then I had to go thru entire install process to get it running again
Example: [https://1drv.ms/i/s!AuFirwug-g8_i6g5BGlLQXgDPxwH8Q](url)
Eample2: [https://1drv.ms/i/s!AuFirwug-g8_iP4WnY2IQfGxpS1uvw](url)

Just need to get 1 machine running Reliably and have several smaller PC's I would like to eventually bring up just nodes on then Share it as a multi-node when I am not using it most of these computers sit Idle on like my CNC Machine & my 3d printer why not help other process some work.  I Have the bandwidth.

This will be week 3 for me working almost day and night to get this to work reliably. My wife is not even a little happy with me. But I'm not one to let some code beet me and I am learning a lot. heck never touched Linux before a week ago have since installed an run 3 different Versions :)  
Anyway I appreciate all you help and time and wish I had the programing knowledge to be more help.

GOD Bless Marvin

pierotofy, 2017-05-30:
Sorry to hear that you are having some issues!

Let's go back a few steps. Your original issue was getting apache + mod_wsgi working?

First I would check that the rest of the application works. If you have followed all of the steps outlined in the README, from the WebODM directory, can you run:

```bash
python manage.py runserver
```

?

marvinlange, 2017-06-05:
PC Matic Super Shield Blowing up Docker Digital image Verification even just disabling it did not help but I removed just super shield now docker works in windows

after finding this post i figured I would try it on my PC Matic
https://social.technet.microsoft.com/Forums/windows/en-US/2df51b2d-18d3-463e-8fe5-3517bfb2950f/vm-failed-to-start-worker-process-the-process-terminated-unexpectedly?forum=win10itprovirt

closed 2017-06-08 by pierotofy