WebODM/WebODM #184: Bad quality images

djamajo, 2017-05-27:
After uploaded images for processing, the output result is not clear. 
I got a Blurry picture.

Here is the log files reported.

`Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0686.JPG | camera: dji fc330 | dimensions: 4000 x 3000 | focal: 3.61 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0423.JPG | camera: dji fc330 | dimensions: 4000 x 3000 | focal: 3.61 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0430.JPG | camera: dji fc330 | dimensions: 4000 x 3000 | focal: 3.61 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0550.JPG | camera: dji fc330 | dimensions: 4000 x 3000 | focal: 3.61 | ccd: None`

this error happens for each image loaded.
Any help

pierotofy, 2017-05-29:
@djamajo does increasing the `--orthophoto-resolution` parameter to something like 60 help?

djamajo, 2017-05-29:
@pierotofy we also notice that the map made from the pictures uploaded that 's not reflect the real image of the area overflew. Images are not in good order and some of the images are not present on the final map.

pierotofy, 2017-06-01:
Hi @djamajo, I'm not sure what you mean by "good order", could you expand further on that?

As far as the missing images, it's probably because the images do not exhibit enough "features" (noticeable objects), common in areas with lots of vegetation, so the reconstruction simply discards them. Try to increase the `--min-num-features` parameter to something like 8000 or 12000 and increase the `--resize-to` parameter as well.

djamajo, 2017-06-01:
Hi @pierotofy, what i mean by missing images is that, after the processing of the images by web odm, some of the images were not part of the final result.

This is the expected result 

[image]

and this is the result we had.

[image]

looking at the two images what can be the real problem that made us have another result

pierotofy, 2017-06-08:
As I replied earlier, have you tried increasing the parameters I recommended?

djamajo, 2017-06-09:
@pierotofy, yes i got the expected result but the texture of the trees are not good.
A part from  `--min-num-features`, `--orthophoto-resolution` and `--resize-` don't you have another parameters to give a good texture of the trees

pierotofy, 2017-06-09:
I'm not sure what you mean by "good texture", perhaps you could include a few screenshots that detail the visual problem with the result?

djamajo, 2017-06-09:
@pierotofy 

After processing with the parameters suggested below , i got this picture A (see picture below). but the quality is still not good compare to picture B processed with another software.
What another features can i use to improve the quality of the images.

Image A

[image]

Image B

[image]

PS: We did the test in High vegetation.

dakotabenjamin, 2017-06-13:
With high vegetation you really need to fly a cross path to get the complete structure of the vegetation. However, you point out some issues with our meshing algorithm. It is a fairly simple Poisson Surface Reconstruction that does not work great for highly vegetated areas. You get those blobs when flattening the model for an orthophoto.

pierotofy, 2017-06-13:
Like @dakotabenjamin said, flying a cross path will yield better results, but I would also use the `--use-25dmesh` option which uses the 2.5D mesh for generating the orthophoto. I would try a large `--mesh-remove-outliers` value (20) and decrease the `--mesh-size` to something like 10000. This should cause the mesh to "flatten". You can get a better understanding of the result if you then open the resulting mesh in the `odm_meshing` folder in MeshLab, you'll see the "blobs" @dakotabenjamin is talking about.

pierotofy, 2017-06-13:
Another thing to try is to use `--use-pmvs`, which uses pmvs instead of opensfm for generating the point cloud. Since the resulting cloud will be much more sparse, it should lead to a better looking orthophoto for this particular dataset.

closed 2017-06-24 by pierotofy