WebODM/WebODM #185: Process exited with code 1

capedrone, 2017-05-29:
I keep running running into the same issue no matter what image set I try to use. Used geotagged images, used none geotagged.. same error. Please could someone suggest what it is that I ma doing wrong or need to change. many thanks 

[WARNING] No coordinates file. Generating coordinates file: /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/opensfm/bundle_r000.out -inputCoordFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_georeferencing/coords.txt -inputFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /var/www/data/86e54013-44f6-49b0-b4ca-34465b00852a/odm_georeferencing/odm_georeferencing_model_geo.txt
Segmentation fault
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 137, in process
    '-logFile {log} -georefFileOutputPath {geo_sys}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139

pierotofy, 2017-05-29:
Hi @capedrone, it's hard to say what the problem could be. Could you share the images you are using so that we can try to reproduce it?

capedrone, 2017-05-29:
Hi Pierotofy: here are just 8 images.. the full set has 89.. please test, I am re-running program now with the same 8 images...
https://www.dropbox.com/sh/dxrdm1ydqpxqpav/AAC3mvZsZm0kJapNJD6QF-Gra?dl=0

Hope the dropbox link work

Kind Regards

capedrone, 2017-05-29:
Just complete the above files and it failed again with : Process exited with code 1
Building objmodel:
	Saving model... Building objmodel:
	Saving model... 	Saving model... 	Saving model... 	Saving model... done.
Whole texturing procedure took: 383.142s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/opensfm/bundle_r000.out -inputCoordFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_georeferencing/coords.txt -inputFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /var/www/data/ca565d54-0e43-4acc-bef9-a3056ba7fdf5/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   lat_frac: 6257096/1 39/1 47772/1000
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 150, in process
    geo_ref.utm_to_latlon(tree.odm_georeferencing_latlon, photo, idx)
  File "/code/opendm/types.py", line 306, in utm_to_latlon
    altitude = abs(int(float(latlon[2]) * 100))
ValueError: cannot convert float NaN to integer

pierotofy, 2017-05-31:
The problem seems to be caused by a fault in the odm_extract_utm program, which generates a Z coordinate with the value `-nan`:

cat odm_georeferencing/coords.txt
```
WGS84 UTM 34S
269069 6257079
-0.8049886768 -8.296736724 0
5.111920786 20.15579969 -nan
27.75191406 20.69522297 -1.270232263
40.58296582 -40.0112147 0
-18.01091157 4.391117054 0
-32.2224741 17.66326714 -1.268045876
16.17286643 -21.0271387 0
-31.41965067 14.37799052 0
```

The program seems to hang-up with:

```
Error in OdmExtractUtm:
Failed to transform coordinates
```

https://github.com/OpenDroneMap/OpenDroneMap/blob/36f7106795ac40510c581e02aca6775e4e1bc0d7/modules/odm_extract_utm/src/UtmExtractor.cpp#L268

I will need to dig into this further. I will post an update when I know more. Thank you for reporting the problem!

pierotofy, 2017-06-08:
Still need to look into this, sorry for the delay.

pierotofy, 2017-06-08:
After applying the changes in https://github.com/OpenDroneMap/OpenDroneMap/pull/596 I was able to process the dataset correctly.

[image]

It was a fault in the odm_utmextract module. Thank you for reporting the issue! I don't think we would have caught it without your help.

As soon as the pull request is merged into OpenDroneMap, wait a day and do a `docker pull opendronemap/node-opendronemap`, then restart WebODM.

capedrone, 2017-06-08:
Fantastic, thanks for the update 👍

closed 2017-06-08 by dakotabenjamin