WebODM/WebODM #188: Web ODM Deployment Idea useing Ubuntu Live

marvinlange, 2017-05-30:
Is their a way to make a Script that will run a complete install of WEB-ODM and have it run imbedded in Ubuntu live Iso  or implement snapcraft.io
to deploy  and update (( https://insights.ubuntu.com/2017/05/30/build-snapcraft-io-gets-your-code-ready-to-distribute-in-minutes/  ))
its just a thought then someone could burn a Bootable DVD image/USB Boot boot from it and it automatically sets up either a Web-Odm server or Processing node 

Just a Thought.

pierotofy, 2017-05-31:
It would be a cool project!

marvinlange, 2017-06-01:
I finally have WEBODM up and running on an a Generic Ubuntu Windows Virtual Machine it has no Personnel Info once I get it finished up Create some slandered accounts I will export The VM and provide a link to it it should make it make it turn-key not as good as UBUNTU Live after 30 Years of Windows well since DOS kind of dating myself  even being a MCSE Learning Linux is no joke :(

marvinlange, 2017-06-01:
Well with my limited Linux Knowledge I did the best thing I can think of to help Windows People.
Made VM running Ubuntu 16.04 With Web Odm Installed
Installed  UI Docker so you can see and Manage Containers with Gui
Added Links in Firefox Toolbar to Web Odm Server, Docker UI
Ubuntu User: odm  Pass: odm
To use 
1: Extract [ubuntu-VM-w-ODM.rar](https://1drv.ms/u/s!AuFirwug-g8_i7BnLnf5gP6pXJOMwQ)  <--11 GIG
2 :open up Hyper-V Manager  Choose  import Virtual Machine Point it extracted Folder and restore to your PC.

pierotofy, 2017-06-01:
That's really cool!

Would you be able to modify the README (open a pull request) to include this information?

marvinlange, 2017-06-02:
Well I think I did it correct 2 you tube videos just to figure out how to make pull request :)

closed 2017-06-08 by pierotofy