WebODM/WebODM #193: Exits ungracefully at end of cleaning depthmaps

KommandorKeen, 2017-06-02:
WebODM says
2017-06-02 06:58:35,227 Cleaning depthmap for image DJI_0019.JPG
2017-06-02 06:58:35,541 Cleaning depthmap for image DJI_0032.JPG
2017-06-02 06:58:36,534 Cleaning depthmap for image DJI_0062.JPG
2017-06-02 06:58:37,348 Cleaning depthmap for image DJI_0075.JPG
2017-06-02 06:58:37,857 Cleaning depthmap for image DJI_0060.JPG

terminal says
INFO Acquiring lock: Task [Task of 2017-06-02T06:40:29.228Z] (3)
webapp        | WARNING Run time of job "background.<locals>.wrapper (trigger: interval[0:00:05], next run at: 2017-06-02 06:59:36 UTC)" was missed by 0:00:01.152061
webapp        | WARNING Run time of job "cleanup_projects (trigger: interval[0:00:15], next run at: 2017-06-02 07:00:06 UTC)" was missed by 0:00:01.025425
webapp        | Execution of job "background.<locals>.wrapper (trigger: interval[0:00:05], next run at: 2017-06-02 07:00:06 UTC)" skipped: maximum number of running instances reached (3)
webapp        | Execution of job "background.<locals>.wrapper (trigger: interval[0:00:05], next run at: 2017-06-02 07:00:11 UTC)" skipped: maximum number of running instances reached (3)
webapp        | WARNING Run time of job "cleanup_projects (trigger: interval[0:00:15], next run at: 2017-06-02 07:00:21 UTC)" was missed by 0:00:07.279826
webapp        | WARNING Run time of job "background.<locals>.wrapper (trigger: interval[0:00:05], next run at: 2017-06-02 07:00:21 UTC)" was missed by 0:00:07.948127
webapp        | Execution of job "background.<locals>.wrapper (trigger: interval[0:00:05], next run at: 2017-06-02 07:00:21 UTC)" skipped: maximum number of running instances reached (3)
webapp        | WARNING Run time of job "cleanup_projects (trigger: interval[0:00:15], next run at: 2017-06-02 07:00:36 UTC)" was missed by 0:00:10.732731
webapp        | [2017-06-02 07:00:25 +0000] [54] [CRITICAL] WORKER TIMEOUT (pid:60)
db            | LOG:  unexpected EOF on client connection with an open transaction
webapp        | [2017-06-02 07:00:53 +0000] [858] [INFO] Booting worker with pid: 858
webapp        | [2017-06-02 07:02:13 +0000] [54] [CRITICAL] WORKER TIMEOUT (pid:858)
webapp        | [2017-06-02 07:02:13 +0000] [858] [INFO] Worker exiting (pid: 858)
webapp        | [2017-06-02 07:02:20 +0000] [859] [INFO] Booting worker with pid: 859
webapp        | [2017-06-02 07:02:51 +0000] [54] [CRITICAL] WORKER TIMEOUT (pid:859)
webapp        | [2017-06-02 07:02:51 +0000] [859] [INFO] Worker exiting (pid: 859)
webapp        | [2017-06-02 07:02:52 +0000] [860] [INFO] Booting worker with pid: 860

KommandorKeen, 2017-06-02:
Final message about 10 mins later....
node-odm-1    | info: About to run: /var/www/scripts/postprocess.sh data/5c841df8-1351-409c-be8e-1e68100b45d9

pierotofy, 2017-06-03:
The process was stuck probably because too many resources were being used; this is normal when running node-odm and webodm on the same machine, if that machine doesn't have enough memory and/or computation power. The solution here is to move the processing node (node-odm) to another machine.

closed 2017-06-03 by pierotofy

KommandorKeen, 2017-06-04:
Thanks,
WebODM is going my Windows box and the grunt work can go on the server.
Will see how that goes.