WebODM/WebODM #194: Task fails with Process exited with code null, no task console output

approxdec, 2017-06-03:
i amusing Docker according to the instructions on the WebODM page. 
i get this response
"Task fails with Process exited with code null, no task console output"
i no i am short on RAM and have not entirely worked that out yet
[image]
and i am unsure how the information in "Common Troubleshooting" applies to me.
your guidance would be much appreciated.

joe martin

pierotofy, 2017-06-03:
Hi @approxdec, thanks for reporting the problem.

It looks like your CPU should support the docker image, so perhaps the problem is caused by something else.

Would you be available to share either publicly or privately the dataset you are using for processing?

approxdec, 2017-06-03:
Thanks very much Piero. I am running a very average laptop with 4gb ram so
I don't want to be a nuisance.

I am in Newcastle nsw Australia which is utc + 10 (midnight now) so I will
get back to you.

Regards
Joe.

Hi @approxdec <https://github.com/approxdec>, thanks for reporting the
problem.

It looks like your CPU should support the docker image, so perhaps the
problem is caused by something else.

Would you be available to share either publicly or privately the dataset
you are using for processing?

approxdec, 2017-06-03:
Piero ...
since my laptop is 7 years old and short on memory i had tried using only 6 files and 10 files since WebODM would not process my 69 image set. this morning i have tried again with 30 images and WebODM has been running for 15 minutes now, so things may be cooking good.

I have no idea how long the processing will take of course but i will get back to you with an result.
as to my original "Task fails with Process exited with code null, no task console output" report
it seems as though i have confused WebODM with my small number image sets in which case
i am hoping that the report may still be useful for the incorrect?? task report.
joe

approxdec, 2017-06-04:
gday ...
i have given 
WebODM 2 tasks - 30 images and 9 images - which it accepted but ran for
9 hours each without result. as i said my laptop is old and slow and short on memory.
[time for a new box] i think you would be chasing ghosts rather than brackets that
aren't closed.

less images than 9 give the "Task fails with Process exited with code null, no task console output".
the image set has produced an object mesh and has been edited in Blender
so i think although my install of WebODM seems okay it can't actually can't do the work.

thanks anyway. i guess i am stuck with that at the moment.
joe

pierotofy, 2017-06-05:
Joe,

would you be available to share either publicly or privately the dataset you are using for processing?

approxdec, 2017-06-05:
That would be generous of you. I am away from computer until 0700 utc
Let me know what you would like and we will do it privately since data set is no mine.

Joe

pierotofy, 2017-06-05:
You can share your images on a file hosting such as google drive or dropbox, then you can send me the link via this form: https://masseranolabs.com/contact/

pierotofy, 2017-06-06:
[image]

The images processed fine, so I can exclude a problem with the dataset. It really seems like your CPU does not like the way we compiled the docker image, although your chipset is not that old (2009) and we should support CPUs that date back to 2006.

The best thing to do here would be to compile OpenDroneMap and node-OpenDroneMap. This is going to require some reading on your part, see the "running natively" sections of https://github.com/OpenDroneMap/OpenDroneMap and https://github.com/OpenDroneMap/node-OpenDroneMap. I wish I could point a simpler solution, but when docker fails we're back to having to compile things from scratch at the moment.

The good news is that at the end of the story, you'll take advantage of all the processing power your CPU has to offer by compiling from scratch. Hope this helps.

approxdec, 2017-06-06:
many thanks for your help and advice.

closed 2017-06-08 by pierotofy