WebODM/WebODM #195: Looking for tasks.json as it is corrupted

KommandorKeen, 2017-06-05:
After a power outage my tasks.json file is reported as corrupted, where do I find it on a docker installation?

Attaching to node-odm-1, db, webapp
db            | LOG:  database system was shut down at 2017-06-05 01:29:42 UTC
db            | LOG:  MultiXact member wraparound protections are now enabled
db            | LOG:  database system is ready to accept connections
webapp        | /bin/bash: /webodm/wait-for-it.sh: /bin/bash: bad interpreter: Text file busy
db            | LOG:  autovacuum launcher started
node-odm-1    | error: Error during startup: Could not load task list. It looks like the data/tasks.json is corrupted (Unexpected end of JSON input). Please manually delete the file and try again.
db            | LOG:  incomplete startup packet
node-odm-1 exited with code 1

pierotofy, 2017-06-05:
Hi @KommandorKeen,

Try to run:

```
docker rm -v $(docker ps -aq --filter name=node-odm-1)
```

You might have to replace node-odm-1 with the name of your node-odm container (see `docker ps -a` for all containers).

You will have to restart WebODM then via `./webodm.sh stop && ./webodm.sh start`.

KommandorKeen, 2017-06-05:
Thanks, That worked.
My next task is to move docker from the boot disk to my 2TB volume as it starting to take up a lot of space.

closed 2017-06-05 by KommandorKeen