WebODM/WebODM #196: Question about image uploads Failing

marvinlange, 2017-06-05:
I have a set of 441 images about 3 meg each Geo-Tagged

If I try to upload them using Web-Odm  it completes the visual upload then just kind of pauses then give me a error that it cannot upload

If I open Node-OpenDroneMap  I can upload all images no problem and starts processing no problem but then Web Odm dos not see the manually uploaded images I assume its assigned some kind of token during upload.,

Specs
Docker Ruining Natively
Running on Windows 10 with 18Gigs of ram  12gig assigned to docker
8 CPU Cores 6 assigned to docker
Dynamic expanding Virtual Drive space with 100 gig Cap
Windows Private Firewall turned Off
Antivirus Disabled

I have tried using
Google Chrome
Firefox
IE Explorer

P.S. I get the Same issue if I try to Use my Ubuntu VM  with the same above specs

pierotofy, 2017-06-05:
Hi @marvinlange, I'm not entirely sure what could be the cause of the problem. I've tried to upload a dataset with 450 images (about 2GB in size) and things worked OK on my Linux Arch machine.

I was going to guess that there was not enough disk space allocated, have you made sure that your docker machine has enough disk space allocated?

marvinlange, 2017-06-05:
so are you saying allocating 100 gig to my VM MobyLinuxVM.vhdx  has no affect on the Docker size allocation that that needs to be adjusted as well ?

marvinlange, 2017-06-05:
Of course the Funny part is that node is in docker and if I upload direct no issue

pierotofy, 2017-06-05:
Mm, then it's probably not a space issue. Allocating space to MobyLinuxVM.vhdx is the right thing to do.

Could you try to:
 - Open the developer console tab in your browser (on Chrome press F12, navigate to the console tab)
 - Upload the images from WebODM
 - What errors do you get? (if any)
 - Also from the network tab, can you copy/paste the network request details that happen to POST /api/projects/<x>/tasks/ ?

Thanks!

marvinlange, 2017-06-06:
ERROR MESSAGE ( Could not upload all files. An error occured. Please try again. )

CONSOLE
main-999f5fa….js:1443 Warning: Accessing PropTypes via the main React package is deprecated. Use the prop-types package from npm instead.
printWarning @ main-999f5fa….js:1443
warning @ main-999f5fa….js:1467
get @ main-999f5fa….js:310
Object.defineProperty.value @ main-999f5fa….js:24628
__webpack_require__ @ main-999f5fa….js:20
Object.defineProperty.value @ main-999f5fa….js:23245
__webpack_require__ @ main-999f5fa….js:20
Object.defineProperty.value @ main-999f5fa….js:23072
__webpack_require__ @ main-999f5fa….js:20
(anonymous) @ main-999f5fa….js:22316
__webpack_require__ @ main-999f5fa….js:20
Object.defineProperty.value @ main-999f5fa….js:22134
__webpack_require__ @ main-999f5fa….js:20
Object.defineProperty.value @ main-999f5fa….js:21985
__webpack_require__ @ main-999f5fa….js:20
Object.defineProperty.value @ main-999f5fa….js:68
__webpack_require__ @ main-999f5fa….js:20
(anonymous) @ main-999f5fa….js:47
__webpack_require__ @ main-999f5fa….js:20
(anonymous) @ main-999f5fa….js:40
(anonymous) @ main-999f5fa….js:43
main-999f5fa….js:63344 Potree 1.5RC
2(index):1 [.Offscreen-For-WebGL-09F626A8]GL ERROR :GL_INVALID_ENUM : GetIntegerv: <- error from previous GL command
main-999f5fa….js:1443 Warning: `value` prop on `textarea` should not be null. Consider using the empty string to clear the component or `undefined` for uncontrolled components.
    in textarea (created by EditProjectDialog)
    in div (created by EditProjectDialog)
    in div (created by EditProjectDialog)
    in form (created by FormDialog)
    in div (created by FormDialog)
    in div (created by FormDialog)
    in div (created by FormDialog)
    in div (created by FormDialog)
    in FormDialog (created by EditProjectDialog)
    in EditProjectDialog (created by ProjectListItem)
    in li (created by ProjectListItem)
    in ProjectListItem (created by ProjectList)
    in ul (created by ProjectList)
    in div (created by Paginator)
    in Paginator (created by ProjectList)
    in div (created by ProjectList)
    in ProjectList (created by projectList)
    in projectList (created by Route)
    in Route (created by Dashboard)
    in div (created by Dashboard)
    in Router (created by BrowserRouter)
    in BrowserRouter (created by Dashboard)
    in Dashboard
printWarning @ main-999f5fa….js:1443
warning @ main-999f5fa….js:1467
handleElement @ main-999f5fa….js:21851
onBeforeUpdateComponent @ main-999f5fa….js:21862
callHook @ main-999f5fa….js:8600
emitEvent @ main-999f5fa….js:8612
onBeforeUpdateComponent @ main-999f5fa….js:8897
receiveComponent @ main-999f5fa….js:8294
updateChildren @ main-999f5fa….js:15278
_reconcilerUpdateChildren @ main-999f5fa….js:14819
_updateChildren @ main-999f5fa….js:14923
updateChildren @ main-999f5fa….js:14910
_updateDOMChildren @ main-999f5fa….js:12140
updateComponent @ main-999f5fa….js:11958
receiveComponent @ main-999f5fa….js:11920
receiveComponent @ main-999f5fa….js:8304
updateChildren @ main-999f5fa….js:15278
_reconcilerUpdateChildren @ main-999f5fa….js:14819
_updateChildren @ main-999f5fa….js:14923
[... 55 lines cut ...]
runBatchedUpdates @ main-999f5fa….js:7923
perform @ main-999f5fa….js:9222
perform @ main-999f5fa….js:9222
perform @ main-999f5fa….js:7862
flushBatchedUpdates @ main-999f5fa….js:7945
closeAll @ main-999f5fa….js:9288
perform @ main-999f5fa….js:9235
batchedUpdates @ main-999f5fa….js:18478
batchedUpdates @ main-999f5fa….js:7870
dispatchEvent @ main-999f5fa….js:18635
main-999f5fa….js:29005 POST http://192.168.2.150:8000/api/projects/9/tasks/ 502 (Bad Gateway)
Dropzone.submitRequest @ main-999f5fa….js:29005
Dropzone.uploadFiles @ main-999f5fa….js:29001
Dropzone.processFiles @ main-999f5fa….js:28794
Dropzone.processQueue @ main-999f5fa….js:28767
(anonymous) @ main-999f5fa….js:28637
[image]

NETWORK TAB
[image]

APPLICATION TAB 
[image]

[image]

marvinlange, 2017-06-06:
<html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx/1.10.3</center>
</body>
</html>

[image]

marvinlange, 2017-06-06:
[image]
Windows IP Configuration

   Host Name . . . . . . . . . . . . : DVR-16CH
   Primary Dns Suffix  . . . . . . . :
   Node Type . . . . . . . . . . . . : Hybrid
   IP Routing Enabled. . . . . . . . : No
   WINS Proxy Enabled. . . . . . . . : No

Ethernet adapter vEthernet (HNS Internal NIC):

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Hyper-V Virtual Ethernet Adapter #3
   Physical Address. . . . . . . . . : 00-15-5D-57-D8-43
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::2d34:11c0:c0cd:bea6%19(Preferred)
   IPv4 Address. . . . . . . . . . . : 172.29.112.1(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :
   DHCPv6 IAID . . . . . . . . . . . : 520099165
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-1D-A1-66-31-E0-CB-4E-2C-4C-84
   DNS Servers . . . . . . . . . . . : fec0:0:0:ffff::1%1
                                       fec0:0:0:ffff::2%1
                                       fec0:0:0:ffff::3%1
   NetBIOS over Tcpip. . . . . . . . : Enabled

Ethernet adapter vEthernet (DockerNAT):

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Hyper-V Virtual Ethernet Adapter #4
   Physical Address. . . . . . . . . : 00-15-5D-70-01-10
   DHCP Enabled. . . . . . . . . . . : No
   Autoconfiguration Enabled . . . . : Yes
   IPv4 Address. . . . . . . . . . . : 10.0.75.1(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . :
   DNS Servers . . . . . . . . . . . : fec0:0:0:ffff::1%1
                                       fec0:0:0:ffff::2%1
                                       fec0:0:0:ffff::3%1
   NetBIOS over Tcpip. . . . . . . . : Enabled

Ethernet adapter Comcast -150:

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Marvell Yukon 88E8056 PCI-E Gigabit Ethernet Controller #2
   Physical Address. . . . . . . . . : E0-CB-4E-2C-4C-83
   DHCP Enabled. . . . . . . . . . . : No
   Autoconfiguration Enabled . . . . : Yes
   IPv6 Address. . . . . . . . . . . : 2603:3021:c01:ca00::2054(Preferred)
   Lease Obtained. . . . . . . . . . : Monday, June 05, 2017 10:49:43
   Lease Expires . . . . . . . . . . : Monday, June 12, 2017 10:37:10
   IPv6 Address. . . . . . . . . . . : 2603:3021:c01:ca00:8122:4a81:9152:594a(Preferred)
   Temporary IPv6 Address. . . . . . : 2603:3021:c01:ca00:1567:ec69:3465:d140(Preferred)
   Link-local IPv6 Address . . . . . : fe80::8122:4a81:9152:594a%17(Preferred)
   IPv4 Address. . . . . . . . . . . : 192.168.2.150(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : fe80::f44b:2aff:fe72:572f%17
                                       192.168.2.1
   DHCPv6 IAID . . . . . . . . . . . : 316721998
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-1D-A1-66-31-E0-CB-4E-2C-4C-84
   DNS Servers . . . . . . . . . . . : 2001:558:feed::1
                                       2001:558:feed::2
                                       192.168.2.1
                                       8.8.8.8
                                       75.75.75.75
   NetBIOS over Tcpip. . . . . . . . : Enabled

Ethernet adapter vEthernet (Virtual Switch (automatic)):

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Hyper-V Virtual Ethernet Adapter #2
   Physical Address. . . . . . . . . : E0-CB-4E-2C-4C-84
   DHCP Enabled. . . . . . . . . . . : No
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::7d54:b244:dcfc:f2f2%4(Preferred)
   Autoconfiguration IPv4 Address. . : 169.254.242.242(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   Default Gateway . . . . . . . . . : 192.168.2.1
   DHCPv6 IAID . . . . . . . . . . . : 534825806
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-1D-A1-66-31-E0-CB-4E-2C-4C-84
   DNS Servers . . . . . . . . . . . : 2001:558:feed::1
                                       2001:558:feed::2
   NetBIOS over Tcpip. . . . . . . . : Enabled

Ethernet adapter Bluetooth Network Connection:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Bluetooth Device (Personal Area Network)
   Physical Address. . . . . . . . . : 00-1A-7D-DA-71-0B
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
PS C:\Users\Marvin\WebODM>

pierotofy, 2017-06-06:
Mm, it might be a problem with nginx, but 502 does not tell much.

Could you try the following:

Start WebODM, then run:

```
docker exec $(docker ps --filter name="webapp" -aq) tail -f /tmp/nginx.error.log
```

Where "webapp" is the name of your webodm container (do a `docker ps` to find out the exact name if it's different than "webapp")

Then try the upload again. Do you see any errors appearing in the console from the `tail` command? Copy/paste them here.

marvinlange, 2017-06-06:
PS C:\Users\Marvin\webodm> docker exec $(docker ps --filter name="webapp" -aq) tail -f /tmp/nginx.error.log
2017/06/06 02:06:16 [error] 53#53: *7 upstream prematurely closed connection while reading response header from upstream, client: 172.18.0.1, server: webodm.localhost, request: "POST /api/projects/12/tasks/ HTTP/1.1", upstream
: "http://unix:/tmp/gunicorn.sock:/api/projects/12/tasks/", host: "192.168.2.150:8000", referrer: "http://192.168.2.150:8000/dashboard/?project_task_open=9&project_task_expanded=3&page=1"
2017/06/06 02:08:15 [error] 53#53: *10 open() "/webodm/build/static/app/bundles/leaflet-src.map" failed (2: No such file or directory), client: 172.18.0.1, server: webodm.localhost, request: "GET /static/app/bundles/leaflet-sr
c.map HTTP/1.1", host: "192.168.2.150:8000"
2017/06/06 02:08:15 [error] 53#53: *10 open() "/webodm/build/static/app/js/vendor/metisMenu.js.map" failed (2: No such file or directory), client: 172.18.0.1, server: webodm.localhost, request: "GET /static/app/js/vendor/metis
Menu.js.map HTTP/1.1", host: "192.168.2.150:8000"
tail: unrecognized file system type 0x794c7630 for ‘/tmp/nginx.error.log’. please report this to bug-coreutils@gnu.org. reverting to polling

marvinlange, 2017-06-06:
PS C:\Users\Marvin> cd webodm
PS C:\Users\Marvin\webodm> docker exec $(docker ps --filter name="webapp" -aq) tail -f /tmp/nginx.error.log
2017/06/06 21:16:54 [error] 54#54: *3 upstream prematurely closed connection while reading response header from upstream, client: 172.18.0.1, server: webodm.localhost, request: "POST /api/projects/13/tasks/ HTTP/1.1", upstream
: "http://unix:/tmp/gunicorn.sock:/api/projects/13/tasks/", host: "192.168.2.150:8000", referrer: "http://192.168.2.150:8000/dashboard/"
2017/06/06 21:19:31 [error] 54#54: *9 upstream prematurely closed connection while reading response header from upstream, client: 172.18.0.1, server: webodm.localhost, request: "POST /api/projects/13/tasks/ HTTP/1.1", upstream
: "http://unix:/tmp/gunicorn.sock:/api/projects/13/tasks/", host: "192.168.2.150:8000", referrer: "http://192.168.2.150:8000/dashboard/"
tail: unrecognized file system type 0x794c7630 for ‘/tmp/nginx.error.log’. please report this to bug-coreutils@gnu.org. reverting to polling
2017/06/06 21:22:48 [error] 54#54: *16 upstream prematurely closed connection while reading response header from upstream, client: 172.18.0.1, server: webodm.localhost, request: "POST /api/projects/12/tasks/ HTTP/1.1", upstrea
m: "http://unix:/tmp/gunicorn.sock:/api/projects/12/tasks/", host: "192.168.2.150:8000", referrer: "http://192.168.2.150:8000/dashboard/?page=1"

marvinlange, 2017-06-07:
pierotofy Is thier a way i can send you a private message I would be more then happy to let you remote into my machine I can temp open up the   Daemon TLS Port or even setup a remote desktop session or I can install VNC

marvinlange, 2017-06-07:
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f0331fa23c8>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpgdbkqj0z.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f0331fa2518>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmprdrb_fvm.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f0331fb3518>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpums1atuq.upload'

marvinlange, 2017-06-07:
Because small groups of files like 40 to 50 work 
I tried making names shorter ie 1,2,3,400 still failed to upload
I also tried reducing 3 meg image  size in half still failed :(

pierotofy, 2017-06-07:
Quick question, how many seconds does it take for the upload of the large datasets? This might be a timeout issue.

pierotofy, 2017-06-07:
What happens if you change the nginx/nginx.conf file at the line:

```
    keepalive_timeout 5;
```

with:

```
    keepalive_timeout 0;
```

And rebuild webodm (./webodm.sh stop && ./webodm.sh rebuild && ./webodm.sh start)?

marvinlange, 2017-06-07:
Tried to pull out as much info as I could  would it helped if I exported this mobyLinux VM  and upload to my MS one-drive for you to download ?

PS C:\Users\Marvin\webodm>  ./webodm.sh rebuild
PS C:\Users\Marvin\webodm> ./webodm.sh start
PS C:\Users\Marvin\webodm> docker exec $(docker ps --filter name="webapp" -aq) tail -f /tmp/nginx.error.log
tail: unrecognized file system type 0x794c7630 for ‘/tmp/nginx.error.log’. please report this to bug-coreutils@gnu.org. reverting to polling
2017/06/07 15:20:38 [error] 53#53: *26 recv() failed (104: Connection reset by peer) while reading response header from upstream, client: 172.18.0.1, server: webodm.localhost, request: "POST /api/projects/16/tasks/ HTTP/1.1",
upstream: "http://unix:/tmp/gunicorn.sock:/api/projects/16/tasks/", host: "192.168.2.150:8000", referrer: "http://192.168.2.150:8000/dashboard/"

attemp #2 

PS C:\Users\Marvin> cd WebODM
PS C:\Users\Marvin\WebODM> ./webodm.sh stop
PS C:\Users\Marvin\WebODM> ./webodm.sh start
PS C:\Users\Marvin\WebODM> docker exec $(docker ps --filter name="webapp" -aq) tail -f /tmp/nginx.error.log
tail: unrecognized file system type 0x794c7630 for ‘/tmp/nginx.error.log’. please report this to bug-coreutils@gnu.org. reverting to polling
2017/06/07 15:33:43 [error] 53#53: *13 open() "/webodm/build/static/app/bundles/leaflet-src.map" failed (2: No such file or directory), client: 172.18.0.1, server: webodm.localhost, request: "GET /static/app/bundles/leaflet-sr
c.map HTTP/1.1", host: "192.168.2.150:8000"
2017/06/07 15:33:43 [error] 53#53: *15 open() "/webodm/build/static/app/js/vendor/metisMenu.js.map" failed (2: No such file or directory), client: 172.18.0.1, server: webodm.localhost, request: "GET /static/app/js/vendor/metis
Menu.js.map HTTP/1.1", host: "192.168.2.150:8000"
2017/06/07 15:35:57 [error] 53#53: *20 recv() failed (104: Connection reset by peer) while reading response header from upstream, client: 172.18.0.1, server: webodm.localhost, request: "POST /api/projects/17/tasks/ HTTP/1.1",
upstream: "http://unix:/tmp/gunicorn.sock:/api/projects/17/tasks/", host: "192.168.2.150:8000", referrer: "http://192.168.2.150:8000/dashboard/"

[image]
[image]
[image]
[image]
[image]
[image]
<html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx/1.10.3</center>
</body>
</html>

marvinlange, 2017-06-07:
And not sure if this makes a Difference But In Docker Kitematic the WebOdm Container Stops when this happens 
[image]
[image]

pierotofy, 2017-06-07:
Thank you for all the info. I'm 90% certain the problem is in the nginx configuration. Did you try changing the line in the nginx.conf file?

marvinlange, 2017-06-07:
Yes I did then I did the Rebuild before I ran it  and I get the same issue on running on pure Linux Ubuntu or windows

pierotofy, 2017-06-07:
I've just pushed some new changes to the repository that increase the timeout on both gunicorn and nginx. https://github.com/OpenDroneMap/WebODM/commit/acd14e9b5bbcb49b5c17071b5f292c57533fe147

Could you try to update your code, rebuild and try one more time?

Thank you!

marvinlange, 2017-06-08:
AWESOME !!!!!!! looks like you did it.
Seeing some wired stuff on my BASH Screen but it went threw and seems to be processing 440 3 meg files :)  and to be clear just in case I deleted my webodm DIR restarted my docker and did a fresh install of WebOdm  

webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpj9zjs89t.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42da860>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpciu4qibd.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42da9b0>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmp_4y5jg0u.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42dab00>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpisb2k_ym.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42dac50>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpe9df5jxo.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42dada0>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmp6lxkcca8.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42daef0>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmptn8kiint.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42e0080>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpokbmwku3.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42e01d0>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmp184fivdo.upload'
webapp        | Exception ignored in: <bound method _TemporaryFileCloser.__del__ of <tempfile._TemporaryFileCloser object at 0x7f89f42e0320>>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 450, in __del__
webapp        |     self.close()
webapp        |   File "/usr/local/lib/python3.5/tempfile.py", line 446, in close
[... 792 lines cut ...]
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpij0ugkbh.upload'
[... cut ...]
webapp        |     unlink(self.name)
webapp        | FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmp3ri61zpn.upload'
node-odm-1    | info: About to run: python /code/run.py --mesh-solver-divide 12 --min-num-features 8000 --use-pmvs --project-path /var/www/data --pmvs-num-cores 7 ff328033-7f5d-4332-8f07-bb3008a79329
webapp        | INFO Acquiring lock: Task [Task of 2017-06-08T00:38:55.887Z] (13)
webapp        | INFO Acquiring lock: Task [Task of 2017-06-08T00:38:55.887Z] (13)

marvinlange, 2017-06-08:
made you a login in case you want to check it out let me know when you see this message so I can delete it and please change your password

http:\\73.216.88.107:8000
user: pierotofy 
pass: webodm#1

pierotofy, 2017-06-08:
Glad it worked!

closed 2017-06-08 by pierotofy