WebODM/WebODM #197: Add panel with disk usage statistics

pierotofy, 2017-06-05:
It would be useful for users to be able to see the amount of disk space remaining available, especially because people are often confused by the difference between the host disk space and the disk space allocated for docker.

pierotofy, 2017-06-06:
We could also display things like the nginx error log, access log and other "debug" information.

KommandorKeen, 2017-06-08:
[image]

On the disk usage issue, any idea what the docker diff is? My webODM directory is big for the size of the data in it, but the diff appears 'suddenly and is 2 x the size.

pierotofy, 2017-06-08:
They are intermediate images. Some of them might be old and could be removed via:

```
docker images -a|grep '^<none>'|tr -s ' '|cut -d' ' -f 3|xargs docker rmi
```

AnthonyCoughlin, 2017-06-09:
@pierotofy, this would be a great addition!

pierotofy, 2020-04-01:
This has been added a while ago.

closed 2020-04-01 by pierotofy