WebODM/WebODM #202: sir images are which folder stored sir how to find sir please help how to change to another folder

boopathit98, 2017-06-09:
sir images are which folder stored sir how to find sir please help how to change to another folder

pierotofy, 2017-06-09:
Could you please provide some more details?

 - Are you using docker?
 - What is your goal? (Rephrase your question)
 - What OS are you using?

jmprovencher, 2017-06-21:
I'm using docker with WEBODM on MacOS and I would like to know where I could locate the file generated in output on my system. (I know I can download the assets in the user interface).

pierotofy, 2017-06-21:
Docker volumes are stored in /Users/<youruser>/Library/Containers/com.docker.docker/Data/com.docker.driver.amd64-linux/Docker.qcow2.

The file is a QEMU QCOW Image (v3). You will need to mount that image as a file system before you can access its files. I'm not entirely familiar with the process for this on MacOS.

If you want access of the files directly from your host, you might want to modify docker-compose to map a directory from the host instead of using named volumes.

https://github.com/OpenDroneMap/WebODM/blob/master/docker-compose.yml#L22

Example:

```
entrypoint: /bin/bash -c "chmod +x /webodm/*.sh && /bin/bash -c \"/webodm/wait-for-it.sh db:5432 -- /webodm/start.sh\""
    volumes:
      - appmedia:/webodm/app/media
```

Change to:

```
entrypoint: /bin/bash -c "chmod +x /webodm/*.sh && /bin/bash -c \"/webodm/wait-for-it.sh db:5432 -- /webodm/start.sh\""
    volumes:
      - /Users/example/appmedia:/webodm/app/media
```

closed 2017-06-23 by pierotofy