WebODM/WebODM #204: Comparison: ODM vs. DroneDeploy

AnthonyCoughlin, 2017-06-10:
Hey ODM development,

I wanted to give you guys some prospective on ODM vs. DroneDeploy.  First off, I primarily use WebODM.  That said, I updated WebODM this morning.  I captured these images this afternoon (10Jun17).  Its a small field house at a sports complex.  I processed the images through WebODM using the default settings and also processed them through DroneDeploy.  See the attached screenshots.  Also below, I've uploaded the images to Google Drive and provided a link if you'd like to inspect. 

Hope this helps further ODM's progress!
Anthony

[image]
[image]
[image]
[image]

https://drive.google.com/open?id=0B4Tf0aIvxRLJai1VM1VSUWV6Rm8

kikislater, 2017-06-11:
Dronedeploy is a pix4d routine batch (pix4d pro on server) with algorithms from Sure photogrammetry processing.

May be you have to tune parameters to get better with odm

AnthonyCoughlin, 2017-06-11:
Ok thanks!  I'm still learning with ODM.  I'll keep testing!  Any suggestions on parameter settings would be appreciated!

KommandorKeen, 2017-06-11:
Increase outlier rejection and increase to 0.8 (or higher) the coherency

KommandorKeen, 2017-06-12:
Here's an idea.
Put the images in a dropbox folder and share.
Then we can all give it a go and work out the best parameterisation.

AnthonyCoughlin, 2017-06-12:
Glad you got the images.  I'm flying an Autel Robotics x-star premium, The camera is made by Sony with the Exmor-R sensor.  I've had limited success when trying to locate the CCD width.  My research has led me to believe that this camera uses cmos and not ccd?  Does this make a difference?

kikislater, 2017-06-12:
Exmor-r is cmos sensor so every cmos sensor are rolling shutter. What's really matter is sensor readout... Some are bad (like sony sensor in gopro or phantom2 and 3) and other better (like rx100, or phantom 4 pro : it's not a global shutter, or parrot sensefly soda). Don't know autel platform but I will look more ...
There is a ton of exmor-r sensor : https://en.m.wikipedia.org/wiki/Exmor

Here it's not an issue. Issue comes from default parameters in odm.

KommandorKeen, 2017-06-12:
Have a look at this.
At a guess the difference is an SOR (statistical outlier rejection) filter of the entire point cloud prior to meshing.

[image]

Issue at the other end of your building due to images being a little far apart.

BUT notice how ODM catches the overhanging roof eaves.

With great configurability comes great responsibility. 

For this kind of work the Pix4D iphone app for data collection is great. it collects at user configurable x and Y distances.

pierotofy, 2017-06-12:
It's more complicated than that; ODM uses a simple Poisson reconstruction for mesh building, Pix4D uses some proprietary implicit surface fitting algorithm that leads to much smoother surfaces (buildings), along with better outlier removal (which is being discussed here https://github.com/OpenDroneMap/OpenDroneMap/issues/583). There are many papers describing such approaches, but none (that I know of) have source code available for use.

It's an area that requires a lot of work to improve. I would work on it myself if I had time, but there are currently more pressing features to be implemented (mission planning, DEMs, volume measurements, etc.).

AnthonyCoughlin, 2017-06-12:
Thanks for all the info.  I plan to continue testing.  @pierotofy looking forward to the new features you mentioned, thanks for all your work sir!

@KommandorKeen any idea if the Pix4D Iphone App will work with Autel Robotics craft?  I plan to do some research this evening.

KommandorKeen, 2017-06-12:
@AnthonyCoughlin, I think the app is DJI focussed.  Simplest option is to set up an interval timer for photo collection and manually thin them after capture. But more images needed.

KommandorKeen, 2017-06-14:
Just discovered Regard3D from MIT (MIT license) currrently evaluating.....
[http://www.regard3d.org/index.php/documentation/tutorial](url)

More interesting are the pointers to other opensource surface consructors
[http://www.gcc.tu-darmstadt.de/home/proj/fssr/](url)

and the opensource CUDA accelerators for point matching
[http://www.gcc.tu-darmstadt.de/home/proj/cuda_surf/index.en.jsp](url)

pierotofy, 2017-06-14:
OpenDroneMap has the option to use PMVS, but OpenSfM does a better job already. Would be interesting to compare MVE.

It would be cool to implement floating scale reconstruction.

SURF is patented, so we cannot include it in OpenDroneMap.

nsicart, 2017-07-15:
You guys are doing a great job! Keep working hard, thaks a lot!

Please if you can make a technical comment on this below, it looks like webodm is doing a bad job.

https://sketchfab.com/models/b2e3edb64ea349f4b7d140869cf18c55#

AnthonyCoughlin, 2017-07-17:
I posted a comment on SketchFab and reiterated the what nsicart and KommandorKeen said.  Hope this clears things up a bit.

https://sketchfab.com/models/b2e3edb64ea349f4b7d140869cf18c55#

FarhadG, 2017-10-09:
@kikislater Is this really true that DroneDeploy uses Pix4D underneath the hood for their photogrammetry service? How can we confirm this?

nickponline, 2018-01-25:
No we doesn't use Pix4D. We have our own photogrammetry pipelines and choose different configurations based on the scene imagery and the scene being reconstructed.

bcjarrett, 2019-07-15:
@nickponline What engine do you use to so SFM?

nickponline, 2019-07-18:
@bcjarrett We wrote our own.

kikislater, 2019-08-27:
Even if it's not pix4d it looks like you use same backend as pix4d : SURE photogrammetry

pierotofy, 2020-04-02:
Let's move this and future comparison conversations over to https://community.opendronemap.org :clinking_glasses: (aside, recent changes to the ODM engine should have improved results quite a bit).

closed 2020-04-02 by pierotofy