WebODM/WebODM #205: Georeferencing appears broken

KommandorKeen, 2017-06-11:
Ran the Bellus test case with both exif gps and and gcp file and get the error below

Is there a typo should ...8a4/gpc/gcp_list.txt 

actually read

....8a4/gcp/gcp_list.txt?

OUTPUT

[DEBUG]   running /code/build/bin/odm_georef -bundleFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/opensfm/bundle_r000.out -imagesPath /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/images_resize -imagesListPath /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/opensfm/list_r000.out -bundleResizedTo 1000 -inputFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/odm_texturing/odm_textured_model.obj -outputFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/opensfm/depthmaps/merged.ply -outputPointCloudFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/odm_georeferencing/odm_georeferenced_model.ply  -logFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/odm_georeferencing/odm_georeferencing_model_geo.txt -gcpFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/gpc/gcp_list.txt -outputCoordFile /var/www/data/5ac4c6d1-fbc3-4688-ae31-05e8f32be8a4/odm_georeferencing/coords.txt
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model

pierotofy, 2017-06-12:
Hi @KommandorKeen, I've just tried to process the bellus dataset and it worked OK using docker node-opendronemap and the default parameters. Maybe we're using a different dataset? Where did you download it?

The GPC/GCP is a typo https://github.com/OpenDroneMap/node-OpenDroneMap/issues/12 but does not influence the result.

KommandorKeen, 2017-06-12:
@pierotofy Straight from the opendronmap page. I will try again.
Thanks

KommandorKeen, 2017-06-13:
The problem is with SFM umber of processes, not the georeferencing per se.
I re-ran the test case straight out of the box it worked, before I had 'speeded it up' with six cores.

closed 2017-06-13 by KommandorKeen

pierotofy, 2017-06-13:
Glad it worked! Can you reproduce consistently the issue if you increase the number of opensfm processes? It might be worth opening a separate issue if you can find a way to reproduce it.