WebODM/WebODM #207: Possible Processing Error due to EXIF Data

kea64, 2017-06-12:
Not that it's a big deal or is a serious issue, but I noticed that processing seems to always fail when I take stills from an SJ Cams SJ6 Legend camera. (I realize this camera does have fisheye and does not geotag, but I have batch corrected photos and geotagged images via Mission Planner which seems to work fine with other cameras. This is more of an experiment than anything else)

The link to a few sample photos is here: [Google Drive](https://drive.google.com/open?id=0B1oldqRO99KMX2dsYThwNHgzdkk)

I'm no expert, but based on the attached task output, it seems like the reason processing fails within the first 30 seconds is due to an error in the image's EXIF data. Keep in mind, I have used other cameras with WebODM and have had success. Really cool stuff going on here. Any help or tips would be welcomed!
[task_output.txt](https://github.com/OpenDroneMap/WebODM/files/1069168/task_output.txt)

pierotofy, 2017-06-12:
Hi @kea64, this looks very similar to #182. Could you read through that and let me know if it applies? It's a known problem.

kea64, 2017-06-13:
Yes, it looks very similar to the issue I'm experiencing now. Looks like I'll need to brush up on my .bat file skills.

closed 2017-06-15 by pierotofy