WebODM/WebODM #208: 3D model using LiDar data as well as multi/highspectral & LiDar data fusion feature

smajida, 2017-06-14:
Hi, as nowadays small lidar scanners are getting cheaper, so that would be great if we could add a new feature of 3D model creation using Lidar data and more importantly to improve the quality of 3D model significanlty using fusion of point cloud from images and lidar data.

I think in order to do so, after creating point cloud from images, we have to read point cloud of Lidar and combine them together and them create a mesh out of resulting point cloud. 
you can consider this repo:
https://github.com/smajida/ros-image_cloud

Also for hyperspectral and lidar fusion:
http://neon-workwithdata.github.io/neon-data-institute-2016/R/neon-data-fusion-R/

as well as:
https://github.com/yeknom/lidar-camera-fusion
https://github.com/smajida/ros-examples
https://github.com/smajida/PulsedLight3dScanner
https://github.com/smajida/fastfusion

pierotofy, 2017-06-14:
Would be interesting to explore this field, is this something you would be able to help us with?

pijemcolu, 2017-06-19:
As of now the simplest way to do this would be to employ an iterative closest point algo (ICP). The algorithm iteratively tries to find a transform between two different point clouds. An example can be found at point cloud library documentation http://pointclouds.org/documentation/tutorials/interactive_icp.php (visualisation at the end). Another code example from openCV is at http://www.morethantechnical.com/2010/06/06/iterative-closest-point-icp-with-opencv-w-code/.

rickeshtn, 2017-09-22:
I would encourage you to take a look at ROS based SLAM algorithms. The list which you can try can be found here. http://www.cvlibs.net/datasets/kitti/eval_odometry.php

pierotofy, 2020-04-02:
It's been 3 years since this issue was first opened. If there's interest in adding this, please open a PR (probably starting from the ODM repository, which would need to include workflows for using lidar data).

closed 2020-04-02 by pierotofy