WebODM/WebODM #211: PMVS vs SFM (georef + point cloud precision)

KommandorKeen, 2017-06-16:
I am having trouble making the georeferencing work with more than 1 SFM core operating, so resorted to PMVS. noticed this difference in the pointcloud LAS files

PMVS
[image]
SFM
[image]

NOte to get georef to work I have to leave the images at their fullsize i.e re-size to 3000x4000.
I am running a job at the moment where I have resized to 1000 and provided suitably scaled geroref points. will let you know how it goes.

pierotofy, 2017-06-16:
This looks similar to https://github.com/OpenDroneMap/OpenDroneMap/issues/522.

dakotabenjamin, 2017-06-16:
This should have been fixed in 0.3... why would it not work for PMVS?

pierotofy, 2017-06-16:
Different input ply format? Not entirely sure. @KommandorKeen could you share the dataset you used to produce this result (along with the parameters used to run the task)?

KommandorKeen, 2017-06-19:
will, do.....soon.  Thanks.

KommandorKeen, 2017-06-20:
Images may be found at...
https://drive.google.com/open?id=0B9GCZwyI5K2yRHl2WVI2djZTekU
Options in WebODM were..
Options: resize-to: 4000, opensfm-processes: 6, use-pmvs: true

I did the full size option as I was /am getting a little doming on biggger areas.

pierotofy, 2017-07-07:
I can't seem to reproduce this.

[image]

@KommandorKeen have you made sure to run the latest version of node-opendronemap?

pierotofy, 2017-07-07:
(Sorry didn't mean to close this)

KommandorKeen, 2017-07-12:
Tried a ./webodm.sh update today and completely trashed my install. :-(
Will have to start from scratch. Looks like I tried to do the update with a disk not mounted so stuff got placed in all sorts of strange places.
.......
later

pierotofy, 2017-07-12:
@KommandorKeen I just pushed an updated in #236 to fix a problem with migrations. Before nuking everything, you should try to update one more time and see if you can bring the program back online. It shouldn't be necessary to start everything from scratch.

KommandorKeen, 2017-07-13:
It does when the problem was the hard drive not liking the cold weather and becoming unusable. Not realising this and trying to update a docker install that wasn’t there. User error.

KommandorKeen, 2017-07-26:
Cannot re-replicate this problem so closing it.

closed 2017-07-26 by KommandorKeen