WebODM/WebODM #213: AgiSoft Vs WebODM

Pratyush1991, 2017-06-16:
Hi Guys,
I am working on multi source images and trying to compare the results with only single source image. In my case I am using images captured from DJI Phantom 3 and Iphone 5s. I have provided below the screenshot of the dense cloud generated from Agisoft and WebOdm. I used default settings in Webodm and while processing in Agisoft settings that I used are as follow:
Align photo- High (Accuracy), Build Dense Cloud- Medium (Quality), Build  Mesh- High (Face count), Build Texture - Generic (Mapping mode), Mosaic (Blending Mode).
Textured Model from Agisoft
[image]

Dense Cloud from Agisoft
[image]

WebODM
[image]
 
In Webodm the roof of the building are missing. So I would like to know what parameters shall I use to improve my results in Webodm such that roof top can be seen.

About the dataset that I have used. I did 3 flight plan in video mode around the building and extracted the images from the video. 2 flight plan from above the building one was in circular video mode and other one was normal photo mode. At last I took photos around the building by holding the DJI Phantom in hand (offline mode) and with Iphone5s.
The photos extracted from video mode didn't have Geo location and sensor information. So I set the 35mm focal length parameters to 20. 
Link to Dataset
https://drive.google.com/drive/folders/0B7tV_P5gZivFMWlQY29mWk1oV1E?usp=sharing

pierotofy, 2017-06-22:
I'm pretty sure the lower density results are due to the fact that the photos extracted from video have no geolocation information.

It would be interesting to see if you get better results by increasing https://github.com/mapillary/OpenSfM/blob/22d70a4df8e5e970f7f8f3739d59139b9225d719/opensfm/config.py#L49 matching_time_neighbors and matching_order_neighbors in opensfm, but we currently don't have a way to pass those parameters from OpenDroneMap.

I'll open an issue.

dakotabenjamin, 2017-06-22:
I'm currently working through the opensfm update. I am imagining a way to pass any of these parameters in the ODM command line. Like `run.py --opensfm-params "lowes_ratio: 0.8"` or something like that.

pierotofy, 2017-06-22:
With that said, you could have an "advanced parameter replacement" mode, where you can manually type infrequently used parameters, which is the `--opensfm-params` approach you described, but I would leave it as an "advanced feature" for experienced users.

dakotabenjamin, 2017-06-22:
Yeah, actually having advanced parameters is a really good idea. Some current params might be considered such as well.

pierotofy, 2017-06-22:
May I also suggest JSON as the format for passing such advanced parameters. --opensfm-params '{"matcher_threshold": 5, ...}'

Pratyush1991, 2017-06-23:
Thanks a lot. Looking forward to the update.
So I have a question.
Is it that the matcher-neighbour and matcher-ratio option inside WebODM same as matching_gps_neighbors and preemptive_lowes_ratio of OpenSFM..??

I tried to update WebODM and got some error as follows:-
Please commit your changes or stash them before you can merge.
error: The following untracked working tree files would be overwritten by merge:
        app/migrations/0003_auto_20170615_1300.py
        app/templatetags/processingnode_extras.py
        docker-compose.build.yml
Please move or remove them before you can merge.
Aborting

Will be glad if you could help. :)

pierotofy, 2017-06-23:
If you've made some changes to the code, you'll need to commit them before you can update your code. Do a search on Google on how to use `git` for merging changes.

pierotofy, 2020-04-02:
If this is still a problem with newer versions of WebODM/ODM, let's move the conversation to https://community.opendronemap.org :+1: thank you!

closed 2020-04-02 by pierotofy