WebODM/WebODM #216: ./webodm.sh start

JUANVALENCIAMOREY, 2017-06-22:
$ ./webodm.sh start
Checking for docker...   OK
Checking for git...   OK
Checking for python...   can't find python! Check that the program is installed and that you have added the proper path to the program to your PATH environment variable before launching WebODM. If you change your PATH environment variable, remember to close and reopen your terminal. https://www.python.org/downloads/

How do i fix it........

pierotofy, 2017-06-23:
Hi Juan,

You need to install Python on your system. The installation process varies depending on what system you are currently using.

JUANVALENCIAMOREY, 2017-06-23:
I always installed it and I do not know why the message appears

pierotofy, 2017-06-23:
What operating system are you on?

JUANVALENCIAMOREY, 2017-06-23:
Windows 10

I have python installed and<https://sourceforge.net/projects/pyscripter/> PyScripter<https://www.google.com.pe/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjgweT379TUAhWC6CYKHa52C70QFggnMAA&url=https%3A%2F%2Fsourceforge.net%2Fprojects%2Fpyscripter%2F&usg=AFQjCNHTjy3qE-tlM0pyprHN6rW_n0REJQ><https://www.google.com.pe/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjgweT379TUAhWC6CYKHa52C70QFggnMAA&url=https%3A%2F%2Fsourceforge.net%2Fprojects%2Fpyscripter%2F&usg=AFQjCNHTjy3qE-tlM0pyprHN6rW_n0REJQ><https://www.google.com.pe/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjgweT379TUAhWC6CYKHa52C70QFggnMAA&url=https%3A%2F%2Fsourceforge.net%2Fprojects%2Fpyscripter%2F&usg=AFQjCNHTjy3qE-tlM0pyprHN6rW_n0REJQ><https://sourceforge.net/projects/pyscripter/>

pierotofy, 2017-06-23:
You need to make sure that the path to your python.exe executable is in your PATH environment variable.

Note that there's also an installer available for Windows, which makes the installation a lot easier: https://webodm.org/installer

closed 2017-06-30 by pierotofy

JUANVALENCIAMOREY, 2017-06-30:
Gracias