WebODM/WebODM #218: Default Login and Password Change

kea64, 2017-06-24:
I've been experimenting with WebODM for about 2 months now, and I'm quite excited to see where all this is going. I've been getting decent results with a variety of different cameras and platforms and have regularly updated to the newest versions to see what all is introduced.

Recently, I installed the newest version of WebODM up to the commit on June 17. I noticed at login, it seems the default username of 'admin' and password of 'admin' has changed. I can no longer login or use WebODM. I already reinstalled WebODM to see if it was a glitch, but the defualt password has changed. 

Has the default login credentials changed or are you guys planning to create a method to create user accounts? If it helps, I'm including a screen shot of the login screen. I'm on Windows 10 Pro at the moment.
[image]

pierotofy, 2017-06-24:
Hi @kea64, there's a new welcome screen that allows a user to choose a username/password the first time WebODM is launched.

There was a bug between June 9th and June 15th where the first account creation mechanism was broken, leading to the creation of a username that can no longer be logged-in. Do you remember creating a username via the welcome screen between those dates?

The procedure to unlock the username is:

1) Start WebODM
2) Run `docker ps`

```
8a563252b60b        pierotofy/webodm_webapp          "/bin/bash -c 'chm..."   8 days ago          Up 8 days           0.0.0.0:8000->8000/tcp    webapp
```

3) Grab the hash (in my case `8a563252b60b`) of the webodm_webapp container
4) Run `docker exec -ti 8a563252b60b bash -c "echo \"from django.contrib.auth.models import User;from django.contrib.auth.hashers import make_password;u=User.objects.get(username='admin');u.password=make_password('admin');u.save()\" | python manage.py shell"` (replace the hash)

The new password will be `admin`.

kea64, 2017-06-24:
Hey @pierotofy 
I actually did not have the option to create a username yet. I think that's mostly to do with not updating between the 9th and 15th. 

I tried your method to change the password and it worked immediately. I'm now logged in and able to process images again. I appreciate the help. I noticed that on the left hand side of the main menu, there is an administration tab with the option to create user accounts. I just created a user account. I'm guessing that's how users were supposed to create accounts, but I after wiping my hard drive, there was no way to add a new user on the welcome screen without having a default login.

closed 2017-06-24 by pierotofy

ParkwayProduction, 2020-10-16:
I am getting invalid credentials, I had two maps on my account that I was working on now I am stuck. I was in accounts today before this error came up. I had two accounts showing, one with my email and the other as unknown, so I erased that account and now I am just stuck. It wont allow me to log in