WebODM/WebODM #219: how to connect the docker-machine in a intranet?

jxjazengjt, 2017-06-25:
if i install a WebODM instance in a computer in a intranet with a router,  how to connect the docker-machine in another computer in the intranet?  The docker-machine is in an ubuntu16.04 environment.

pierotofy, 2017-06-26:
Hi @jxjazengjt, you should be able to access WebODM from a computer in the same network, just use http://<ipofmachine>:8000. Is that not the case?

jxjazengjt, 2017-06-27:
Now it's ok. In the browser side  of webodm, I want add a new node in the LAN, just connect the node installed with a docker  via its IP and port(8000)?  Does the new node  need to be specially configued ?

pierotofy, 2017-06-27:
You need to run (on the new computer that acts as a node):

```
docker run -p -d 3000:3000 opendronemap/node-opendronemap
```

This is going to be on port 3000, not 8000.

jxjazengjt, 2017-06-27:
Hi @pierotofy, I just got some prompts:  "docker: Invalid containerPort: -d".  Furthermore, how to know  the new added node works or not ?

pierotofy, 2017-06-27:
Ups, my fault, this is the correct command:

```
docker run -d -p 3000:3000 opendronemap/node-opendronemap
```

To test the node, simply open a web browser to http://ipaddressofnode:3000. If it loads a web page, it's working.

I also encourage you to read the docker docs (https://docs.docker.com/) for learning how to use docker, it will save you some headaches in the long run.

jxjazengjt, 2017-06-28:
@pierotofy,  thanks!  Now it's ok. But does a task use  only one node ? Is it possible to  simultaneously use two or more nodes to improve the efficiency?

pierotofy, 2017-06-28:
It's possible to process two separate tasks on two separate nodes. We are working on using two nodes simultaneously for a single task, but it's a feature that is not available yet.

pierotofy, 2017-06-29:
https://smathermather.com/2017/03/08/scaling-opendronemap-necessary-and-fun-next-steps/

closed 2017-06-29 by pierotofy