WebODM/WebODM #220: Tasks / Image uploads

thuhol, 2017-06-26:
I have WebODM manager 1.0.0
When I'm going to add a Tasks or Image uploads in the site http://ip-address:8000/admin/app/task/add/
"Server Error (500)"

My console config:
Detected Windows 10 with Hyper-V support
Initializing console...
SET DOCKER_TLS_VERIFY=1
SET DOCKER_HOST=tcp://192.168.99.100:2376
SET DOCKER_CERT_PATH=C:\docker-webodm\machines\webodm
SET DOCKER_MACHINE_NAME=webodm
SET COMPOSE_CONVERT_WINDOWS_PATHS=true
REM Run this command to configure your shell: 
REM 	@FOR /f "tokens=*" %i IN ('docker-machine --storage-path C:\\docker-webodm env webodm') DO @%i
Setup env variable: DOCKER_TLS_VERIFY=1
Setup env variable: DOCKER_HOST=tcp://192.168.99.100:2376
Setup env variable: DOCKER_CERT_PATH=C:\docker-webodm\machines\webodm
Setup env variable: DOCKER_MACHINE_NAME=webodm
Setup env variable: COMPOSE_CONVERT_WINDOWS_PATHS=true
Client:
 Version:      17.05.0-ce
 API version:  1.29
 Go version:   go1.7.5
 Git commit:   89658be
 Built:        Fri May  5 15:36:11 2017
 OS/Arch:      windows/amd64

Server:
 Version:      17.05.0-ce
 API version:  1.29 (minimum version 1.12)
 Go version:   go1.7.5
 Git commit:   89658be
 Built:        Thu May  4 21:43:09 2017
 OS/Arch:      linux/amd64
 Experimental: false
192.168.99.100
We are ready :)
WebODM has started. You can close this window now. WebODM will continue running.

pierotofy, 2017-06-26:
Hi @thuhol, generally Tasks are created by uploading images directly from the dashboard, not from the admin interface. In the dashboard use the "Upload Images and GCP" button to create a new task.

This looks like a bug in WebODM administration though and should be fixed. Thanks for reporting it!

http://localhost:8000/admin/app/task/add/

[image]

pierotofy, 2017-07-10:
Fixed in #230

closed 2017-07-10 by pierotofy