WebODM/WebODM #222: Setting overlap by more than 70%

djamajo, 2017-06-28:
This link: https://blog.dronedeploy.com/slantrange-dronedeploy-part-i-51d151193774 , gives how to set the overlap but there are no clear guidlines on what make a good overlap percentage.
Based on the image attach (see image below ) , who can help set a good overlap percentage to more than 70 % ?

[image]

Thanks,
Djama

pierotofy, 2017-06-29:
Those numbers work well for use in SLANTRANGE, which is the product referenced in your link.

I don't think there are clear guidelines on what are the optimal overlap settings; altitude, camera, ground speed and other factors would throw off any fixed rules.

As far as ODM is concerned, more overlap = denser reconstruction, but also more data and longer processing time.

Perhaps we should be discussing the problems you might be currently having with the default overlap settings, and make suggestions from there.

dakotabenjamin, 2017-06-29:
also need to consider the complexity of the subject. buildings and trees need more overlap

djamajo, 2017-06-29:
@pierotofy and @dakotabenjamin 
I conduct more test in environnement where the vegetation is very high, and the density of the trees are very important.
The purpose of my work is to use WebOdm to bring the best of the reconstruction of the images collected during the fly made with Fantom 4.
The issue is how to use DroneDeploy Mobile APP to set setting for SideLap, FrontLap , Altitude, Speed to avec more than 70% Overlap.
Based on what i read , we need to set high SideLap and FrontLap, also Fly at high altitude and fly slower to have a good quality image with more features.

According to the recommandation made by Drone Deploy, i set this setting for my fly: FrontLap 90%, SideLap 75% , altitude 80% and the speed of 10m/s.
Is it a good setting for area with dense vegetation, where the trees are very collapse and tall ?
See image bellow

[image]

juvinski, 2017-06-29:
Hi Djalma,
What camera are you using ?
What is the resolution and what sd card ?

djamajo, 2017-06-30:
@juvinski here are the specifications of the camera for PHANTOM 4
DJI does not list a model of camera for the Phantom 4, although its specs are almost identical to that of the Zenmuse X3, the standard camera found on the DJI Osmo handheld stabilizer and DJI Inspire 1 drone.

Sensor	1/2.3” Effective pixels:12 M
Lens	FOV ( Field Of View ) 94° 20 mm ( 35 mm format equivalent ) f/2.8 focus at ∞
ISO Range	100 - 3200 ( video ) 100 - 1600 ( photo )
Electronic Shutter Speed	8 s to 1/8000 s
Max Image Size	4000×3000
Still Photography Modes	Single shot
For more details , please check this link below: 
http://www.drone-world.com/dji-phantom-4-camera/

KommandorKeen, 2017-07-03:
CCD Width 6.24mm
Focal Length 3.61mm

For flying over trees I fly a grid at at least 3 x the tree height with the camera pointing 15 degrees forward. This allows a look under the canopy.

The Pix4D app for iphone is great for planning and executing DJI missions.

djamajo, 2017-07-03:
@KommandorKeen  i will try that one too

closed 2017-08-18 by pierotofy