WebODM/WebODM #223: simulating high speed battlefield aircraft

werkineh, 2017-06-29:
Hi, there I am MSc. computer networking student. by now I am doing my thesis on supersonic speed aircraft, speed range from 1mach to 5 mach. I need to set the future waypoints of the craft, unfortunately, I can't get any model that used to model this and can be integrated with any of networking simulator. can I use the simulation data from Ardudrone map to any of network simulator. if I can which one is the best simulator which accepts the mission based data from this simulation like ns2, opnet, qualnet, which one?can i simulate networking part with ArduDrone map simulator when they communicate? can I make the plane to communicate their mission plan each other on fly? please help me I have two mouths to submit the thesis

pierotofy, 2017-06-29:
How is this relevant to WebODM?

closed 2017-06-29 by pierotofy

dakotabenjamin, 2017-06-29:
This is the most interesting issue i've seen in a while, but I'm not sure what it has to do with WebODM. Maybe you could try the ArduPilot forums or one of their software issue queues? 

https://github.com/ardupilot