WebODM/WebODM #224: The user and passwd

robeson1010, 2017-06-30:
When I install a new WEBODM and login first. It need the user and passwd, I trid admin admin. It shows'‘Invalid credentials. Note that both fields are case-sensitive.’

How to login？

pierotofy, 2017-06-30:
Hi, you should be able to create the first admin account by visiting http://yourip:8000/welcome/ (that page should load automatically the first time you visit http://yourip:8000).

Did the welcome screen not show up?

robeson1010, 2017-06-30:
yes, i can't load the welcome page and it will jump to the login page automatic.

pierotofy, 2017-06-30:
Did the welcome screen show up the first time you opened WebODM? The reason you are getting redirected is because an admin account exists in the database.

robeson1010, 2017-06-30:
Never see that page~~ I tried to rebuild the docker image and update them. But still can't login

robeson1010, 2017-06-30:
[image]
like this

pierotofy, 2017-06-30:
That's really odd. You can follow these steps to reset the password on the admin account. https://github.com/OpenDroneMap/WebODM/issues/218

?

When did you first install WebODM? Is this a fresh installation or have you installed WebODM previously (and recently updated)?

robeson1010, 2017-06-30:
Yes, I'm sure that my first time to install the WEBODM.   I have tried that methods #218 and it shows:

> Traceback (most recent call last):
  File "manage.py", line 22, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.5/site-packages/django/core/management/__init__.py", line 363, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.5/site-packages/django/core/management/__init__.py", line 355, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.5/site-packages/django/core/management/base.py", line 283, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.5/site-packages/django/core/management/base.py", line 330, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.5/site-packages/django/core/management/commands/shell.py", line 101, in handle
    exec(sys.stdin.read())
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.5/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/usr/local/lib/python3.5/site-packages/django/db/models/query.py", line 380, in get
    self.model._meta.object_name
django.contrib.auth.models.DoesNotExist: User matching query does not exist

[image]
[image]

pierotofy, 2017-06-30:
Mm, the mystery deepens!

What is the output of:

```
docker exec -ti 8a563252b60b bash -c "echo \"from django.contrib.auth.models import User;print(User.objects.filter(is_superuser=True).first().username)\" | python manage.py shell"
```

?

Replace 8a56... with the hash of your container.

robeson1010, 2017-06-30:
green512 ~~~~

robeson1010, 2017-06-30:
Thanks again. I know maybe my colleague used this before and I change  the passwd~

closed 2017-06-30 by robeson1010