WebODM/WebODM #226: Distributed Processing across multiple nodes

markwbrown, 2017-07-04:
Per the 42/C4, I am opening an issue to get the ball rolling on distributed processing. That is, splitting a task across multiple nodes. See [Scaling OpenDroneMap, necessary (and fun!) next steps](https://smathermather.com/2017/03/08/scaling-opendronemap-necessary-and-fun-next-steps/). 

I've seen some talk about Celery, but it looks like efforts to incorporate that were abandoned. Let's hear some proposals for this challenge.

pierotofy, 2017-07-06:
I know Paulinus has started some work on this in this branch: https://github.com/OpenDroneMap/OpenDroneMap/compare/master...paulinus:feature-split-dataset

For sure the work needs to be implemented in ODM before we can add it into WebODM.

pierotofy, 2019-12-03:
This is now available via split-merge (see https://docs.opendronemap.org/large.html and ClusterODM).

closed 2019-12-03 by pierotofy